WebODM/WebODM #760: ERROR 6: zoom_level > 22 not supported NodeMICMAC

rumenmitrev, 2019-12-05:
I will post the issue as I opened in NodeMICMAC repository ,  because cant figure where is the problem exactly.

pierotofy, 2019-12-05:
Hi @rumenmitrev,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-12-08 07:10) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2019-12-05:
Thanks @rumenmitrev :+1: I think this might need to be fixed on the NodeMICMAC side.

closed 2019-12-05 by pierotofy