WebODM/WebODM #761: WebODM is stuck during startup

mohsinalich, 2019-12-07:
WebODM is stuck on starting. 
It was working fine till 24 hours ago.
Can you please advise what could be the issue.
Here is the copy of console

**************************************************************************************
C:\WebODM Manager>cd WebODM/ 
C:\WebODM Manager\WebODM>git pull origin master 
From https://github.com/OpenDroneMap/WebODM
 * branch            master     -> FETCH_HEAD
Already up to date.
C:\WebODM Manager\WebODM>docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml down --remove-orphans 
Removing network webodm_default
Network webodm_default not found.
C:\WebODM Manager\WebODM>docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml pull 
Pulling broker   ... 
Pulling db       ... 
Pulling worker   ... 
Pulling node-odm ... 
Pulling webapp   ... 
[2A[2K
Pulling node-odm ... pulling from opendronemap/nodeodm
[2B[2A[2K
Pulling node-odm ... digest: sha256:b46bda1e357695ac7b...
[2B[2A[2K
Pulling node-odm ... status: image is up to date for o...
[2B[2A[2K
Pulling node-odm ... [32mdone[0m
[2B[3A[2K
Pulling worker   ... pulling from opendronemap/webodm_...
[3B[3A[2K
Pulling worker   ... digest: sha256:ddacbbdd10f98fdf6b...
[3B[3A[2K
Pulling worker   ... status: image is up to date for o...
[3B[3A[2K
Pulling worker   ... [32mdone[0m
[3B[5A[2K
Pulling broker   ... pulling from library/redis
[5B[1A[2K
Pulling webapp   ... pulling from opendronemap/webodm_...
[1B[1A[2K
Pulling webapp   ... digest: sha256:ddacbbdd10f98fdf6b...
[1B[1A[2K
Pulling webapp   ... status: image is up to date for o...
[1B[1A[2K
Pulling webapp   ... [32mdone[0m
[1B[4A[2K
Pulling db       ... pulling from opendronemap/webodm_db
[4B[4A[2K
Pulling db       ... digest: sha256:437e8772fddeb7e740...
[4B[4A[2K
Pulling db       ... status: image is up to date for o...
[4B[4A[2K
Pulling db       ... [32mdone[0m
[4B[5A[2K
Pulling broker   ... digest: sha256:1eedfc017b0cd3e232...
[5B[5A[2K
Pulling broker   ... status: image is up to date for r...
[5B[5A[2K
Pulling broker   ... [32mdone[0m
C:\WebODM Manager\WebODM>exit 0 
Update successful :) Press "Start WebODM" to continue.
[5B
C:\WebODM Manager>cd WebODM/ 
C:\WebODM Manager\WebODM>docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start   || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up 
Starting db       ... 
Starting node-odm ... 
Starting broker   ... 
Starting worker   ... 
Starting webapp   ... 
[4A[2K
Starting node-odm ... [31mfailed[0m
[4B[3A[2K
Starting broker   ... [31mfailed[0m
[3B[5A[2K
Starting db       ... [31mfailed[0m
[5B[2A[2K
Starting worker   ... [31mfailed[0m
[2B[1A[2K
Starting webapp   ... [31mfailed[0m
[1BNo containers to start
Creating network "webodm_default" with the default driver
Creating webodm_node-odm_1 ... 
Creating db                ... 
Creating broker            ... 
[2A[2K
Creating db                ... [32mdone[0m
[2B[1A[2K
Creating broker            ... [32mdone[0m
[1BCreating worker            ... 
[4A[2K
Creating webodm_node-odm_1 ... [32mdone[0m
[4B[1A[2K
Creating worker            ... [32mdone[0m
[1BCreating webapp            ... 
[1A[2K
Attaching to db, broker, webodm_node-odm_1, worker, webapp
Creating webapp            ... [32mdone[0m
[36mdb          |[0m LOG:  database system was shut down at 2019-12-07 10:16:52 UTC
[36mdb          |[0m LOG:  MultiXact member wraparound protections are now enabled
[36mdb          |[0m LOG:  database system is ready to accept connections
[36mdb          |[0m LOG:  autovacuum launcher started
[33mbroker      |[0m 1:C 07 Dec 2019 10:17:36.962 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
[33mbroker      |[0m 1:C 07 Dec 2019 10:17:36.962 # Redis version=5.0.7, bits=64, commit=00000000, modified=0, pid=1, just started
[33mbroker      |[0m 1:C 07 Dec 2019 10:17:36.962 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
[33mbroker      |[0m 1:M 07 Dec 2019 10:17:36.963 * Running mode=standalone, port=6379.
[33mbroker      |[0m 1:M 07 Dec 2019 10:17:36.963 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
[33mbroker      |[0m 1:M 07 Dec 2019 10:17:36.963 # Server initialized
[33mbroker      |[0m 1:M 07 Dec 2019 10:17:36.963 # WARNING you have Transparent Huge Pages (THP) support enabled in your kernel. This will create latency and memory usage issues with Redis. To fix this issue run the command 'echo never > /sys/kernel/mm/transparent_hugepage/enabled' as root, and add it to your /etc/rc.local in order to retain the setting after a reboot. Redis must be restarted after THP is disabled.
[33mbroker      |[0m 1:M 07 Dec 2019 10:17:36.963 * Ready to accept connections
[32mnode-odm_1  |[0m info: Authentication using NoTokenRequired
[32mnode-odm_1  |[0m info: Listening on 0.0.0.0:6367 UDP for progress updates
[32mnode-odm_1  |[0m info: No tasks dump found
[32mnode-odm_1  |[0m info: Checking for orphaned directories to be removed...
[32mnode-odm_1  |[0m info: Server has started on port 3000
[35mworker      |[0m Postgres is up - executing command
[35mworker      |[0m wait-for-it.sh: waiting for broker:6379 without a timeout
[35mworker      |[0m wait-for-it.sh: broker:6379 is available after 0 seconds
[35mworker      |[0m wait-for-it.sh: waiting for webapp:8000 without a timeout
[31mwebapp      |[0m Postgres is up - executing command
[31mwebapp      |[0m wait-for-it.sh: waiting for broker:6379 without a timeout
[31mwebapp      |[0m wait-for-it.sh: broker:6379 is available after 0 seconds
[31mwebapp      |[0m [92m
[31mwebapp      |[0m  _       __     __    ____  ____  __  ___
[31mwebapp      |[0m | |     / /__  / /_  / __ \/ __ \/  |/  /
[31mwebapp      |[0m | | /| / / _ \/ __ \/ / / / / / / /|_/ / 
[31mwebapp      |[0m | |/ |/ /  __/ /_/ / /_/ / /_/ / /  / /  
[31mwebapp      |[0m |__/|__/\___/_.___/\____/_____/_/  /_/   
[31mwebapp      |[0m 
[31mwebapp      |[0m [39m
[31mwebapp      |[0m Checking python version... 3.x, good!
[31mwebapp      |[0m Checking GDAL version... GDAL 2.4.3, released 2019/10/28, excellent!
[31mwebapp      |[0m Running migrations
[31mwebapp      |[0m Generated secret key
[31mwebapp      |[0m INFO Initializing GRASS engine using /usr/bin/grass76
[31mwebapp      |[0m Operations to perform:
[31mwebapp      |[0m   Apply all migrations: admin, app, auth, contenttypes, guardian, nodeodm, sessions
[31mwebapp      |[0m Running migrations:
[31mwebapp      |[0m   Applying app.0030_assure_cogeo...=====================================================================
[31mwebapp      |[0m Migrating TIFFs to Cloud Optimized GeoTIFFs, this might take a while!
[31mwebapp      |[0m =====================================================================
[31mwebapp      |[0m Optimizing /webodm/app/media/project/5/task/9baa5b1b-0810-4814-8232-e5ea381fe881/assets/odm_orthophoto/odm_orthophoto.tif
[31mwebapp      |[0m Optimizing /webodm/app/media/project/12/task/b34e28a6-e821-4794-b96f-70000e39ad47/assets/odm_orthophoto/odm_orthophoto.tif
[31mwebapp      |[0m Optimizing /webodm/app/media/project/26/task/6961316f-3ed0-442d-af4c-f419af94047c/assets/odm_orthophoto/odm_orthophoto.tif
[31mwebapp      |[0m The following warnings were found:
[31mwebapp      |[0m - The file is greater than 512xH or 512xW, it is recommended to include internal overviews
[31mwebapp      |[0m 
[31mwebapp      |[0m INFO Optimizing /webodm/app/media/project/26/task/6961316f-3ed0-442d-af4c-f419af94047c/assets/odm_orthophoto/odm_orthophoto.tif as Cloud Optimized GeoTIFF

*********************************************************************************

[image]

Thanks

pierotofy, 2019-12-07:
Hi @mohsinalich,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-12-10 05:35) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

mohsinalich, 2019-12-07:
What's your browser and operating system? 
Chrome 78 on Windows 10

What is the problem?
WebODM is stuck on starting.
It was working fine till 24 hours ago.

What should be the expected behavior?
It should startup

How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Here is the copy of console

C:\WebODM Manager>cd WebODM/
C:\WebODM Manager\WebODM>git pull origin master
From https://github.com/OpenDroneMap/WebODM

branch master -> FETCH_HEAD
Already up to date.
C:\WebODM Manager\WebODM>docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml down --remove-orphans
Removing network webodm_default
Network webodm_default not found.
C:\WebODM Manager\WebODM>docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml pull
Pulling broker ...
Pulling db ...
Pulling worker ...
Pulling node-odm ...
Pulling webapp ...
�[2A�[2K
Pulling node-odm ... pulling from opendronemap/nodeodm
�[2B�[2A�[2K
Pulling node-odm ... digest: sha256:b46bda1e357695ac7b...
�[2B�[2A�[2K
Pulling node-odm ... status: image is up to date for o...
�[2B�[2A�[2K
Pulling node-odm ... �[32mdone�[0m
�[2B�[3A�[2K
Pulling worker ... pulling from opendronemap/webodm_...
�[3B�[3A�[2K
Pulling worker ... digest: sha256:ddacbbdd10f98fdf6b...
�[3B�[3A�[2K
Pulling worker ... status: image is up to date for o...
�[3B�[3A�[2K
Pulling worker ... �[32mdone�[0m
�[3B�[5A�[2K
Pulling broker ... pulling from library/redis
�[5B�[1A�[2K
Pulling webapp ... pulling from opendronemap/webodm_...
�[1B�[1A�[2K
Pulling webapp ... digest: sha256:ddacbbdd10f98fdf6b...
�[1B�[1A�[2K
Pulling webapp ... status: image is up to date for o...
�[1B�[1A�[2K
Pulling webapp ... �[32mdone�[0m
�[1B�[4A�[2K
Pulling db ... pulling from opendronemap/webodm_db
�[4B�[4A�[2K
Pulling db ... digest: sha256:437e8772fddeb7e740...
�[4B�[4A�[2K
Pulling db ... status: image is up to date for o...
�[4B�[4A�[2K
Pulling db ... �[32mdone�[0m
�[4B�[5A�[2K
Pulling broker ... digest: sha256:1eedfc017b0cd3e232...
�[5B�[5A�[2K
Pulling broker ... status: image is up to date for r...
�[5B�[5A�[2K
Pulling broker ... �[32mdone�[0m
C:\WebODM Manager\WebODM>exit 0
Update successful :) Press "Start WebODM" to continue.
[... 63 lines cut ...]
�[31mwebapp |�[0m | | /| / / _ / __ / / / / / / / /|/ /
�[31mwebapp |�[0m | |/ |/ / __/ // / // / // / / / /
�[31mwebapp |�[0m |/|/_/./_/__// //
�[31mwebapp |�[0m
�[31mwebapp |�[0m �[39m
�[31mwebapp |�[0m Checking python version... 3.x, good!
�[31mwebapp |�[0m Checking GDAL version... GDAL 2.4.3, released 2019/10/28, excellent!
�[31mwebapp |�[0m Running migrations
�[31mwebapp |�[0m Generated secret key
�[31mwebapp |�[0m INFO Initializing GRASS engine using /usr/bin/grass76
�[31mwebapp |�[0m Operations to perform:
�[31mwebapp |�[0m Apply all migrations: admin, app, auth, contenttypes, guardian, nodeodm, sessions
�[31mwebapp |�[0m Running migrations:
�[31mwebapp |�[0m Applying app.0030_assure_cogeo...=====================================================================
�[31mwebapp |�[0m Migrating TIFFs to Cloud Optimized GeoTIFFs, this might take a while!
�[31mwebapp |�[0m =====================================================================
�[31mwebapp |�[0m Optimizing /webodm/app/media/project/5/task/9baa5b1b-0810-4814-8232-e5ea381fe881/assets/odm_orthophoto/odm_orthophoto.tif
�[31mwebapp |�[0m Optimizing /webodm/app/media/project/12/task/b34e28a6-e821-4794-b96f-70000e39ad47/assets/odm_orthophoto/odm_orthophoto.tif
�[31mwebapp |�[0m Optimizing /webodm/app/media/project/26/task/6961316f-3ed0-442d-af4c-f419af94047c/assets/odm_orthophoto/odm_orthophoto.tif
�[31mwebapp |�[0m The following warnings were found:
�[31mwebapp |�[0m - The file is greater than 512xH or 512xW, it is recommended to include internal overviews
�[31mwebapp |�[0m
�[31mwebapp |�[0m INFO Optimizing /webodm/app/media/project/26/task/6961316f-3ed0-442d-af4c-f419af94047c/assets/odm_orthophoto/odm_orthophoto.tif as Cloud Optimized GeoTIFF

[image]

pierotofy, 2019-12-07:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: This issue is related to WebODM Manager, not WebODM. Thank you! :+1:

closed 2019-12-07 by pierotofy