WebODM/WebODM #764: Volume calculation fails to complete.

dmbworld, 2019-12-09:
### What's your browser and operating system?
Firefox 71 on Windows 10
WebODM running on Ubuntu 19.04

### What is the problem?

Volume measurement in 2D Ortho view fails.
Contours appear to be broken with no contours displayed

### What should be the expected behavior?

Normally the volume should be calculated and displayed.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

This is the error displayed for Volume:
Could not execute GRASS script /webodm/plugins/measure/calc_volume.grass from /webodm/app/media/tmp/tmpf9edh3ds_grass_engine: WARNING: Searched for a web browser, but none found Starting GRASS GIS... Creating new GRASS GIS location <location>... Traceback (most recent call last): File "/usr/bin/grass76", line 2211, in <module> main() File "/usr/bin/grass76", line 2124, in main geofile=params.geofile, create_new=True) File "/usr/bin/grass76", line 1024, in set_mapset create_location(gisdbase, location_name, geofile) File "/usr/bin/grass76", line 833, in create_location filename=geostring) File "/usr/lib/grass76/etc/python/grass/script/core.py", line 1538, in create_location encoding='utf-8', mode='w') File "/usr/lib/python2.7/codecs.py", line 898, in open file = __builtin__.open(filename, mode, buffering) IOError: [Errno 2] No such file or directory: '/webodm/app/media/tmp/tmpf9edh3ds_grass_engine/location/PERMANENT/MYNAME

Here is a screen dump:
[image]

This is the error for Contours:
{"readyState":4,"responseText":"<!DOCTYPE html>\n<html lang=\"en\">\n<head>\n \n\n <meta charset=\"UTF-8\">\n <meta http-equiv=\"X-UA-Compatible\" content=\"IE=edge\">\n <meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">\n <meta name=\"author\" content=\"WebODM Contributors\">\n <link rel=\"icon\" type=\"image/png\" href=\"\" sizes=\"32x32\">\n\n <link rel=\"stylesheet\" type=\"text/css\" href=\"/static/app/css/bootstrap.min.css\" />\n\n <link href=\"/static/app/css/font-awesome.min.css\" rel=\"stylesheet\" type=\"text/css\">\n <link href=\"/static/app/css/metisMenu.min.css\" rel=\"stylesheet\">\n <link href=\"/static/app/css/sb-admin-2.css\" rel=\"stylesheet\">\n \n\n <script src=\"/static/app/js/vendor/modernizr-2.8.3.min.js\"></script>\n <script src=\"/static/app/js/vendor/es6-shim.min.js\"></script>\n <script src=\"/static/app/js/vendor/jquery-1.11.2.min.js\"></script>\n <script src=\"/static/app/js/vendor/system.js\"></script>\n\n \n <script type=\"text/javascript\" src=\"/static/app/bundles/main-420078e0836704d3f62c.js\" ></script>\n<link type=\"text/css\" href=\"/static/app/bundles/css/main-420078e0836704d3f62c.css\" rel=\"stylesheet\" />\n\n \n <script src='/plugins/fullscreen/main.js'>

And a screen dump with the additional Contours error:
[image]

pierotofy, 2019-12-09:
Can confirm, this is a problem. Thanks for the report @dmbworld !

closed 2019-12-09 by pierotofy

pierotofy, 2019-12-10:
This should have been fixed. Could you update and confirm @dmbworld ? 🙏

dmbworld, 2019-12-10:
Updated and tested, all is now well, thanks @pierotofy !

PR #765 (2019-12-09): Fixed contours, volume measurement, upgraded GRASS GIS to 7.8
Should fix #764, #736