WebODM/WebODM #766: Volume Calculation is too large and Ortho image pixelated

dmbworld, 2019-12-10:
### What's your browser and operating system?
Firefox 71 on Windows 10
WebODM server running on Ubuntu 19.04
whatismybrowser.com/w/DMLG6JE

### What is the problem?
Volume calculation in current version is inaccurate and on average four to five times greater than in older version of WebODM which is actually the correct calculation. Additionally in the newer vesion of WebODM Ortho view the image is pixelated and scaled down.

### What should be the expected behavior?
Correct volume expected and not so pixelated image

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Older WebODM Image with volume calculation:
[image]

Newer WebODM Image with volume calculation:
[image]

As can be seen in the newer version the image is pixelated, smaller in file size and shows an inflated volume of ariund four times the actual size in the older version.

pierotofy, 2019-12-10:
Thanks for the report @dmbworld 

I think this is a bug due to the fact that the DSM/DTM is now in web mercator projection (EPSG:3857), before we used UTM, and GRASS is now calculating values in that projection, which is expected to be off by:

```(1 / cos (latitude)) ^ 2`

For your area I think that's a factor of `5.20`, which explains the different results.

Still not sure about the lower resolution, but can confirm that results in the web viewer look more pixelated compared to opening them in QGIS.

dmbworld, 2019-12-10:
Just seen the merge and downloaded what appeared to be an update, calculation is still exhibiting the similar multiple too great.

pierotofy, 2019-12-10:
Docker images haven't been updated yet; try again in a few hours https://hub.docker.com/repository/docker/opendronemap/webodm_webapp

dmbworld, 2019-12-10:
Ok, thanks @pierotofy

dmbworld, 2019-12-10:
Yep, it has now updated correctly, sorry about preempting the update. Volumes now look as they should.
The extra pixelation in the view is a little unusual and the TIFF files are smaller in this task after the update run on the existing TIF files. Could this simply be due to a fault in the viewer as you suggested? The older version did not allow such a large zoom as this version does, could that indicate something?

dmbworld, 2019-12-10:
Is it better to move this pixelation feature/reduced image quality to a new issue as I have some more information that may help?

pierotofy, 2019-12-10:
It's OK to leave the pixelation feature here, I've opened a PR that should address this. The problem was both related to the choice of sampling algorithm for tiling and allowed zoom levels in leaflet. :+1:

dmbworld, 2019-12-10:
Thanks @pierotofy!

PR #768 (2019-12-10): Allow upsampling and downsampling past computed zoom levels
Also nearest resampling for orthophoto leads to sharper results.

Should fix #766
closed 2019-12-10 by pierotofy