WebODM/WebODM #770: Surface Model colour key reversed and tile lines maybe be more visible

dmbworld, 2019-12-10:
### What's your browser and operating system? (Copy/paste the output of 
Firefrox 71 on Windows 10
WebODM running on Ubuntu 19.04
whatismybrowser.com/w/HLFQAT9

### What is the problem?
The surface model displays the colour key in reverse order compared to earlier version of WebODM

### What should be the expected behavior?
High levels -> RED
Low Levels -> BLUE

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Original Colour Key:
[image]

New Colour Key:
[image]

dmbworld, 2019-12-10:
There are clearly more options available with the coloring and now I'm not sure if this is an issue or operating as intended?

pierotofy, 2019-12-10:
Ah, good catch! Yes we can color elevation models however we want using any of the palettes from matplotlib https://matplotlib.org/3.1.0/tutorials/colors/colormaps.html

I've reversed the colors in #769

pierotofy, 2019-12-10:
Tile lines should also improve with #769

closed 2020-02-07 by pierotofy