WebODM/WebODM #777: NVDI and ENDVI from NGB cameras

Miguelistan, 2019-12-17:
Hi guys,

After the last improvements on ODM for crops, I noticed something very important is missing.
Since the NGB format is the most popular format for cameras for agricultural use, all the indexes obtained from them are vital for people working with drones in crops. 

Sadly there is no possibility today of getting NDVI from NGB cameras in ODM. I tried to add it myself, but since I am not a developer neither familiar with coding, I failed.  Here you can see the thread:

https://community.opendronemap.org/t/adding-ndvi-and-endvi-for-ngb-cameras/3290/5

Is there anybody that can help out with this? We farmers will thank you a lot.

pierotofy, 2019-12-17:
Hi @Miguelistan,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-12-20 12:40) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Miguelistan, 2019-12-17:
What's your browser and operating system?
Chrome and Windows

What is the problem?
It is not possible to process NDVI neither ENDVI with NGB cameras (most popular format for agricultural cameras)

What should be the expected behavior?
ODM should be able to process both indexes by using the following formulas:
- NDVI is (N-B)/(N+B)
- ENDVI is ((N+G) - (2B)) / ((N+G) + (2B))

How can we reproduce this?
You can follow this thread: 
https://community.opendronemap.org/t/adding-ndvi-and-endvi-for-ngb-cameras/3290/5

pierotofy, 2019-12-18:
Interesting, OK for ENDVI: `((NIR + Green) - (2 x Blue)) / ((NIR + Green) + (2 x Blue))`

NDVI technically uses the `Red` band, is this a variation (we can call it `NDVI (Blue)`)?

Miguelistan, 2019-12-18:
The traditional NDVI formula using Red and NIR was made for old film cameras due to Kodak film availability for that purpose.
Digital cameras respond different than film.  The most important thing is that the camera has near infrared and visible bands in separate channels. Once we have the NIR isolated in a color channel, you can perform NDVI type measurements: the blue and green channels see visible light while the red sees the NIR.  So the NDVI formula can be re-written as ( NIR-BLUE)/(NIR+BLUE)
From my naive point of view, modifying a bit what you have done, you can get the NDVI (Blue) and ENDVI from NGB cameras, that are the most popular for agriculture.

pierotofy, 2019-12-18:
Also interesting read on blue vs. red. https://publiclab.org/notes/nedhorning/10-30-2013/red-vs-blue-filters-for-ndvi

Miguelistan, 2019-12-18:
It is interesting, but this is from 2013 and he has a DIY camera (who knows what kind of blue filter is he using).

I use a Mapir NGB, and the results are extremely good for measuring different vigor levels and non photosynthetic materials. Attached you can find an example processed at dronedeploy.com

It would be amazing to have the possibility of doing the same with ODM.

[image]

smathermather, 2019-12-18:
Ned is not wrong about blue vs. red. He is talking about his specific use cases and filters, but the principles are general.

The challenge in adapting cameras for creating NDVI, including the MAPIR NGB is that unless you are going to purpose build the chip, the simplest way to enable an NIR channel on a commodity chip is to remove the IR filter, add a red blocking filter, and then your red channel will be sensitive to NIR, your green and blue much less so, and you can do some band ratios like NDVI. Doing your band ratios with green won't be as sensitive and doing them with blue, and as you are using a red-blocker and adapting your red channel to detect IR, a red channel NDVI isn't an option.

It is a compromise to do build an NGB sensor, and it is better, when possible, to use an NRG camera. However, for most purposes, the NGB compromise is just fine, and gives good results like the ones you show.

smathermather, 2019-12-18:
Also, Piero has already implemented this and merged it in: https://github.com/OpenDroneMap/WebODM/pull/778 :)

closed 2019-12-18 by smathermather