WebODM/WebODM #781: Allow orthophoto, plant health and surface models to be displayed simultaneously

pierotofy, 2019-12-20:
Good improvements would include:

 * Allow a user to add orthophoto, plant health and DEMs at the same time
 * Better transparency controls in layers panel
 * Allow Z re-ordering of layers

Saijin-Naib, 2020-03-02:
Possibility to swipe/peel between model results would also be awesome.

Happyashbunny, 2024-02-04:
Can you please describe the issue further, I'm not able to comprehend exactly what you're asking

Saijin-Naib, 2024-02-04:
Yep, pretty much exactly what Lupinopr linked. 

@danbjoseph is helping upgrade my fork of UAVArena to include functionality we would like to integrate into WebODM's viewer, if you would like to see/test that and help us there to start?

Saijin-Naib, 2024-02-04:
https://github.com/Saijin-Naib/UAVArena

This is where we are working on stuff.

Once we get the Project finished and working nicely, we want to upstream to the OpenDroneMap Org UAVArena repo, and then try integrating these with WebODM's View Map 2D View.

Any help would be so, so appreciated.

I am way out of my depth, and I am leaning very hard on Dan's free time and generosity to implement all this.

danbjoseph, 2024-02-05:
saying "helping" is being very generous. I am also out my depth 🤕 and was able to do a few front-end enhancements modifying the javascript and Leaflet code. i've poked around the more advanced stuff @Saijin-Naib envisions, but haven't had the time to learn what i need to make it happen. happy for any suggestions to what i've contributed. and i am not far enough along on anything else to stop someone from working on an issue as anything i've finished is already pushed.

Saijin-Naib, 2024-02-05:
I dunno, Dan. I would have accomplished literally nothing without you 🤷

Seems plenty helpful, haha

pierotofy, 2024-11-27:
Fund now available to sponsor this issue: https://fund.webodm.org/#/fund/7

PR #1597 (2025-02-08): Simultaneous map items display and swipe view
Closes #781
closed 2025-02-08 by pierotofy