WebODM/WebODM #783: Process Hung Up Without Error Info

mbos96, 2019-12-29:
### What's your browser and operating system?
I am using Chrome 79 on Windows 10, however, the node is on a different PC: 
[image] 

### What is the problem?
The process seems to get 'hung up' every time it nearly finishes without given error messages. I am trying out a few different settings, which might be causing the problem, but I cannot determine what part is problematic. See also the attached screenshot: 
[image]. 
Additionally, the RAM usage usually decreases to a stable level when this point is reached, and runtime does not progress anymore. That is, the time continues when looking at the interface, but when reloading the page it comes back down to 03:55:08 for the first process, 04:22:56 for the second process, etc.

### What should be the expected behavior?
The expected behavior would be either the process continues and is eventually successful such that resulting files are available for download, or alternatively some sort of error message.

### How can we reproduce this?
I suspect the dataset is not causing problems as I've had successful runs with it. I fear either the setup with an outside node or the settings profile is causing this problem, which I hope is depicted sufficiently in the screenshots.

mbos96, 2020-01-07:
As of yet, I cannot seem to fix the problem, the process seems to get stuck when postprocessing. More specifically, the last thing it logs is related to the Cloud Optimized GeoTIFF:
[image]
After this log, the time got stuck again, as described above. The only logs received after this message came from 'broker' and 'webapp'.

pierotofy, 2020-01-07:
Thanks for the report @mbos96 :hand: does this happen with all datasets or just a specific one?

mbos96, 2020-01-07:
Not exactly all, but I do have this problem with multiple datasets. The particular one, of which I've shared the log above, I was able to successfully run with the same settings by decreasing the number of cameras <50.

pierotofy, 2020-01-08:
Mm, if it gets stuck at optimizing the cloud optimized geotiff, perhaps there's a problem there; if you list your running containers:

`docker ps`

You should be able to find the hash id of your webodm_webapp container:

```
88aa5234a2fa        opendronemap/webodm_webapp         "command"   11 days ago
```

(The hash would be `88aa....`)

Then if you run the `docker cp` command to extract the odm_orthophoto.tif file:

`docker cp 88aa:/webodm/app/media/[......]/odm_orthophoto/odm_orthophoto.tif /home/<myuser>`

That should copy the `odm_orthophoto.tif` file from the container to your home folder. If you can get us a copy of the `odm_orthophoto.tif` file that gets stuck during conversion we can try to troubleshoot it?

mbos96, 2020-01-11:
Sorry, I cannot seem to recreate the exact same problem. Currently, it either finishes perfectly fine, or the process gets 'hung' earlier. To be specific, I've had the same issue where time does not continue, so to speak, when the task output is at 'Reconstructing all views...' twice in the last few days.

smathermather, 2020-01-11:
Depthmap resolution of 2000 is pretty high. Are your images a minimum of 4000 pixels in the smallest dimension?

Also, if you are not turning off resize, a depthmap resolution of 2000 isn't going to work well, as the default resize resolution is 2400, in which case you wouldn't want a depthmap resolution above 1200.

mbos96, 2020-01-11:
Ah sorry, resize is in fact turned off, but that's below on the WebODM interface. However, the images are 3648 pixels in the smallest dimension, so I should probably still lower depthmap resolution it to something like 1500?

smathermather, 2020-01-13:
Yup. I wouldn't go above ~1800. Also, while using larger depthmap resolutions can be useful, it can also introduce noise, so use those increases cautiously. Often with 24MP images (6000x4000 pixels), I often don't go above 1200 for Bayor style sensors (most sensors available).

mbos96, 2020-01-15:
Thanks for this information! Since a few days, I actually have a suspicion of what may cause this problem, but I cannot really check easily. Anyways, to me, it seems that changing either the project name, project description, or task name, after starting the pipeline, may have caused this to occur.

pierotofy, 2020-01-15:
Oh, interesting. If you find a way to reproduce this consistently, please let us know how; it might be a bug.

elkaps, 2020-02-05:
I'm having a similar issue:
[image]

A different set of images works fine but this one fails everytime since I updated the docker images.

pierotofy, 2020-02-05:
Could you post the output of:

```docker logs worker```

?

pierotofy, 2020-02-05:
Mm, seems to be caused by 7zip hanging. I've reverted some of the latest changes which seem to be causing this. https://github.com/OpenDroneMap/NodeODM/pull/110

smathermather, 2020-05-02:
Good news @elkaps. Closing!

closed 2020-05-02 by smathermather