WebODM/WebODM #784: Certbot: error: unrecognized arguments: --tls-sni-01-port 8000

rligocki, 2019-12-30:
A few days ago I installed WebODM (based on docker) and without ssl, everything worked. Then I want to add HTTPS over SSL and I get this error:
> webapp      | certbot: error: unrecognized arguments: --tls-sni-01-port 8000

I searched for this parameter and found out that certbot no longer support --tls-sni-01. Is there any fix for this? I checkout v1.2.0 in webODM folder (./webodm.sh update) and rebuilded all docker containers (./webodm.sh rebuild). Command to run ssl was:
> ./webodm.sh restart --ssl --hostname abc.domain.com  --media-dir /yyy/xxx/WebODM/data/

pierotofy, 2019-12-30:
Hi @rligocki,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-01-02 03:40) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

rligocki, 2019-12-30:
What is the problem?

I am trying to get webODM instance running on my Ubuntu Server 18.04. I have installed docker and docker-compose. Then I clone webODM repository. Using ./webodm.sh start command i get webODM running instance without SSL. To have access over HTTPS it need to have lets encrypt certificate. With command (./webodm.sh restart --ssl --hostname abc.domain.com --media-dir /yyy/xxx/WebODM/data/) i get error in logs:

> webapp      | Postgres is up - executing command
> webapp      | wait-for-it.sh: waiting for broker:6379 without a timeout
> webapp      | wait-for-it.sh: broker:6379 is available after 0 seconds
> webapp      | 
> webapp      |  _       __     __    ____  ____  __  ___
> webapp      | | |     / /__  / /_  / __ \/ __ \/  |/  /
> webapp      | | | /| / / _ \/ __ \/ / / / / / / /|_/ / 
> webapp      | | |/ |/ /  __/ /_/ / /_/ / /_/ / /  / /  
> webapp      | |__/|__/\___/_.___/\____/_____/_/  /_/   
> webapp      | 
> webapp      | 
> webapp      | Checking python version... 3.x, good!
> webapp      | Checking GDAL version... GDAL 2.4.3, released 2019/10/28, excellent!
> webapp      | Running migrations
> webapp      | Generated secret key
> webapp      | INFO Initializing GRASS engine using /usr/bin/grass78
> webapp      | Operations to perform:
> webapp      |   Apply all migrations: admin, app, auth, contenttypes, guardian, nodeodm, sessions
> webapp      | Running migrations:
> webapp      |   No migrations to apply.
> webapp      | INFO Initializing GRASS engine using /usr/bin/grass78
> webapp      | Checking for celery...   OK
> webapp      | Scheduler is not running.
> webapp      | Generating nginx configurations from templates...
> webapp      | - nginx/nginx-ssl.conf
> webapp      | - nginx/nginx.conf
> webapp      | Launching letsencrypt-autogen.sh
> webapp      | celery beat v4.1.0 (latentcall) is starting.
> webapp      | usage: 
> webapp      |   certbot [SUBCOMMAND] [options] [-d DOMAIN] [-d DOMAIN] ...
> webapp      | 
> webapp      | Certbot can obtain and install HTTPS/TLS/SSL certificates.  By default,
> webapp      | it will attempt to use a webserver both for obtaining and installing the
> webapp      | certificate. 
> webapp      | certbot: error: unrecognized arguments: --tls-sni-01-port 8000
> webapp      | WARN: We couldn't automatically generate the SSL certificate. Review the console log. WebODM will likely be inaccessible.
> webapp      | Using nginx SSL configuration
> webapp      | nginx: [warn] the "ssl" directive is deprecated, use the "listen ... ssl" directive instead in /webodm/nginx/nginx-ssl.conf:43
> webapp      | nginx: [emerg] cannot load certificate "/webodm/nginx/ssl/cert.pem": BIO_new_file() failed (SSL: error:02001002:system library:fopen:No such file or directory:fopen('/webodm/nginx/ssl/cert.pem','r') error:2006D080:BIO routines:BIO_new_file:no such file)
> webapp      | 
> webapp      | Trying to establish communication...
> webapp      | 
> webapp      | Something doesn't look right! ¯\_(ツ)_/¯
> webapp      | The server returned a status code of 000 when we tried to reach it.
> webapp      | ==========================
> webapp      | 
> webapp      | Check if WebODM is running, maybe we tried to reach it too soon.

What should be the expected behavior?

After running command (./webodm.sh restart --ssl --hostname abc.domain.com --media-dir /yyy/xxx/WebODM/data/) there should be instance of WebODM accessible over HTTPS with valid lets encrypt certificate. 

How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

```
git clone https://github.com/OpenDroneMap/WebODM.git
cd WebODM
./webodm.sh restart --ssl --hostname abc.domain.com --media-dir /yyy/xxx/WebODM/data/
```

pierotofy, 2019-12-30:
Thanks for the report, it's probably the same as #756. Hoping someone will open a pull request to fix it.

pierotofy, 2019-12-31:
Should be fixed in #785. Please test and see if it's fixed. If not, please re-open this issue?

Thanks.

closed 2019-12-31 by pierotofy