WebODM/WebODM #787: Unusual Output of ODM with GCP file generated from GCP Interface of WebODM

SifatMd, 2020-01-04:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions, comments and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Type answer here]

### What is the problem?

[Type answer here]

### What should be the expected behavior?

[Type answer here]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[Type answer here]

pierotofy, 2020-01-04:
Duplicate of https://github.com/OpenDroneMap/ODM/issues/1060. Please do not open multiple issues!

closed 2020-01-04 by pierotofy

SifatMd, 2020-01-04:
Okay, I will keep that in mind. Thank you.