WebODM/WebODM #788: Add ability to export/preview contours in feet

pierotofy, 2020-01-09:
Proposed in https://community.opendronemap.org/t/contours-in-feet/2804/4

syxs, 2022-03-03:
I just found this will switch the 3d viewer into displaying via Feet with this JS. I use it in a web browser and you could change it back to meters with m instead of ft in the 2nd part. Just not sure how to do this with the current layout of webODM

`viewer.setLengthUnitAndDisplayUnit('m','ft')`

syxs, 2022-03-17:
I have a work around that should be usable now. Its not the best but it will get the job done. 
In app/static/app/js/vendor/potree/build/potree/sidebar.html Just after 
`<div id="scene_export"></div>`

I added these lines:

`<li style="margin: 10px 0px">
					<label for="optLengthUnit-button" class="pv-select-label"><span data-i18n="appearance.length_unit">Length unit</span></label>
						<a href="#" id="mm" onclick="viewer.setLengthUnitAndDisplayUnit('m','m');">Meter</a>
						<a href="#" id="mft" onclick="viewer.setLengthUnitAndDisplayUnit('m','ft');">Feet</a>
						<a href="#" id="min" onclick="viewer.setLengthUnitAndDisplayUnit('m','in');">Inch</a> 
					</li>`

This allows you to select Scene and select what is displayed in the length without changing the base unit of meters.

pierotofy, 2022-03-17:
Would you be interested in opening a pull request with these changes @syxs ? They might be useful to others (and this way you don't have to keep a separate fork).

pierotofy, 2024-05-28:
We now fully support imperial units. :clinking_glasses:

closed 2024-05-28 by pierotofy