WebODM/WebODM #791: Task Completed, but no pointcloud in potree viewer

HeadLessHUN, 2020-01-19:
# What's your browser and operating system?

chrome

### What is the problem?

couldn't load point cloud, in 3D viewer there is no issue in the log from entwine converter. But the dense pointcloud can be downloaded and viewed in other softwares, in potree as well, if i convert it later... (~180 mb laz file)
similar like #738 

### What should be the expected behavior?

Can be viewed by integrated potree viewer

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

i used High resoltion settings with 521 images which are resized to 712px

Richardsonmapper, 2020-02-04:
I am not very experienced at this so please bear with me... 
After using WebODM successfully for the past 6-7 months I recently did an update last week which introduced the Plant Health updates etc and we are now experiencing the same issue as is described above i.e. the 3D viewer in WebODM is now not working and is giving an error tab, see below attachment.

 "Could not load point cloud. This task doesn't seem to have one. Try processing the task again."
[image]

I can however download the textured model .obj file from the WebODM interface and open it in Microsoft 3D viewer but there  appears to be an anomaly in the central portion of the model. i.e. a bulge in the Wetland / Pan which has been flown (See below attachment). Not sure if this is a WebODM processing or image capture issue though. There definitely appears to be a problem with the WebODM 3D viewer interface though. 

[image]

pierotofy, 2020-04-02:
Are the images for this dataset available somewhere so we can try to reproduce this?

Richardsonmapper, 2020-04-05:
Hi Piero, The images are 2.6gb in size zipped, what is the best way to share these with you? RegardsJohn Richardson Sent from Mail for Windows 10 From: Piero ToffaninSent: Friday, 03 April 2020 1:22 amTo: OpenDroneMap/WebODMCc: Richardsonmapper; CommentSubject: Re: [OpenDroneMap/WebODM] Task Completed, but no pointcloud in potree viewer (#791) Are the images for this dataset available somewhere so we can try to reproduce this?

pierotofy, 2020-04-05:
Probably via Dropbox or Google Drive.

Richardsonmapper, 2020-04-09:
Hi Piero,

Attached below is the Google Drive download link, apologies for the delay
but I have been experiencing some problems with regards to my internet
connection and have been battling to upload them.

https://drive.google.com/file/d/1xhvm1gm67Ol2-Fb6nOTNRtDarMaKRzC_/view?usp=sharing

Regards

pierotofy, 2020-04-09:
Does this problem happen with all datasets or just this one? For example, does this one display a point cloud? https://github.com/pierotofy/drone_dataset_brighton_beach/

What's the disk space available? (Check from the Diagnostic panel)

pierotofy, 2020-04-09:
The dataset itself is OK.

[image]

pierotofy, 2020-04-09:
My guess is that you might have run out of disk space while processing the data?

Richardsonmapper, 2020-04-11:
Hi Piero, I have just checked the disk size - 723gb of space available with 13.5gb of memory available. The above dataset just hangs during processing after leaving it running for 2 days. 

I am busy testing the Brighton Beach dataset at the moment and will revert back to you shortly.

Richardsonmapper, 2020-04-11:
[image]

Brighton Beach appears to work fine?

pierotofy, 2020-04-11:
Perhaps you don't have sufficient memory to process it. Does it work if you use a processing node from https://webodm.net ?

pierotofy, 2021-02-05:
Closing this, newer versions of the software should have fixed this. Feel free to re-open if the issue persists? :pray:

closed 2021-02-05 by pierotofy