WebODM/WebODM #792: Idea: Implement raster vision module/plugin for AI object detection

pierotofy, 2020-01-21:
Could be useful for things like tree counting, building detection, cars counting, etc.

Would like to get some feedback on this from users as to whether this would be a useful feature, possible use cases aside from those listed, etc.

smathermather, 2020-01-21:
It's quite possible in certain deciduous forests flown at high resolution , we can count trees in the forest and potentially extract crown size as well:
[image]

We just laid out a project to play with this idea, but hadn't yet chosen a framework (though rastervision seems ideal). Integration into WebODM would be icing.

ChipwizBen, 2020-01-21:
This would be very useful. I've seen livestock counted in the same way.

coreysnipes, 2020-01-21:
This would be a very powerful addition.  I was looking into this a bit recently, also.  These may not all have suitable licensing or tech stack, but other candidates for consideration:

- https://www.tensorflow.org
- https://pjreddie.com/darknet/yolo/
- https://github.com/facebookresearch/detectron2
- https://luminoth.ai/  (deprecated I think)

timotay07, 2020-01-21:
This would be amazing.  I have thought of implementing computer vision to find GCP targets that are QR codes to simplify surveying workflow processes.

smathermather, 2020-01-23:
@timotay07 -- maybe if you have the time, you could look into integrating this as an extension into WebODM: https://github.com/dronemapper-io/aruco-geobits

JamesLeonDufour, 2020-01-23:
Definately a good ideas , counting shelter for estimating displacement population etc...
https://www.arcgis.com/apps/Cascade/index.html?appid=5fdca0f47f1a46498002f39894fcd26f

pierotofy, 2022-06-25:
I think this could be implemented, but requires developing our own smaller library for AI detection; rastervision is just very difficult to install (in terms of dependencies).

SabraHashemi, 2022-12-14:
I was thinking about this right now, why this amazing software dose not have object detection, 
have you done anything already?
I look forward to this feature

AffiTheCreator, 2023-02-04:
So I agree with @SabraHashemi  and so I started developing my own plugin for ODM to use my own or any AI model to perform basicly any type of detection/recognition ... I plan on developing this further and create a whole ecosystem around it but that can be discuss later :)

I'm currently at the stage where I have all the code for a first test and have created a simple interface using what has present in other plugins. I have also reached a stage where I can build the plugin successfully  with ./webodm.sh start --dev but that is all that happens unfortunately, hence my little trip to this thread. 

The only thing modification i did has to coreplugins is there something else missing? Because the plugin does not show in dev tools in the browser under plugins like the others.

SabraHashemi, 2023-02-08:
@AffiTheCreator 
can you join the community of this project?
 there they will answer your quests, and I support your work and efforts

AffiTheCreator, 2023-02-08:
@SabraHashemi I have created an account in the community forum. Are you aware of any current thread where we can discuss this or should i create a new one? 

Also @smathermather would it be possible to share what you have on this subject to compare to what I have currently?

SabraHashemi, 2023-02-09:
> @SabraHashemi I have created an account in the community forum. Are you aware of any current thread where we can discuss this or should i create a new one?
> 
> Also @smathermather would it be possible to share what you have on this subject to compare to what I have currently?

We should create a thread on object detection plugin and also change detection(object based) is a good idea 

Currently I’m working on live videos but its not for this app, after that i will join to contribute to object detection plugin 

Create a thread and i will support and ask the developers to join and help

AffiTheCreator, 2023-02-16:
@SabraHashemi I'll start the thread and share some of my ideas, if you could provide feedback on what you think is best I would appreciate.

pierotofy, 2025-01-20:
Started working on a lib at https://github.com/uav4geo/GeoDeep

First output on 50mb GeoTIFF: 

[image]

```
time python test.py 

real	0m1.988s
user	0m3.143s
sys	0m0.127s
```

pierotofy, 2025-01-20:
If anyone has datasets for tree counts / crowns detection, feel free to share, I will probably start testing a model for that task.

PR #1589 (2025-01-24): AI Object Detection
Adds support for performing AI Object Detection.

Closes #792 

[image]
closed 2025-01-24 by pierotofy