WebODM/WebODM #794: OSM Digitize export link doesnt work, need to add tiles/{zoom}/{x}/{y}.png

JamesLeonDufour, 2020-01-22:
The export code to have the back ground map in Open Street Map doesn't work as it is,

eg: http://xxxxxx/api/projects/7/tasks/81f2d788-9d8b-4133-99ac-484b4a063873/orthophoto/ 

 We need to add : tiles/{zoom}/{x}/{y}.png at the end like this

http://xxxxxx/api/projects/7/tasks/81f2d788-9d8b-4133-99ac-484b4a063873/orthophoto/**tiles/{zoom}/{x}/{y}.png**

Any easy way to fix this ? Otherwise this plugin is amazing @pierotofy

Cheers

pierotofy, 2020-01-22:
Hi @JamesLeonDufour,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-01-24 21:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

JamesLeonDufour, 2020-01-22:
What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Chrome 79 on Windows 10

What is the problem?
The export code to have the back ground map in Open Street Map doesn't work as it is,

eg: http://xxxxxx/api/projects/7/tasks/81f2d788-9d8b-4133-99ac-484b4a063873/orthophoto/

What should be the expected behavior?

We need to add : tiles/{zoom}/{x}/{y}.png at the end like this

http://xxxxxx/api/projects/7/tasks/81f2d788-9d8b-4133-99ac-484b4a063873/orthophoto/**tiles/{zoom}/{x}/{y}.png**

How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Just used the OSM Digitize plugin, and the tiles didnt work so I had to find the fix.

pierotofy, 2020-01-22:
Duplicate of #795

closed 2020-01-22 by pierotofy