WebODM/WebODM #795: OSM Digitize export link doesnt work, need to add tiles/{zoom}/{x}/{y}.png

JamesLeonDufour, 2020-01-22:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 79 on Windows 10

### What is the problem?

The export code to have the back ground map in Open Street Map doesn't work as it is,

eg: http://xxxxxx/api/projects/7/tasks/81f2d788-9d8b-4133-99ac-484b4a063873/orthophoto/

### What should be the expected behavior?

We need to add : tiles/{zoom}/{x}/{y}.png at the end like this

http://xxxxxx/api/projects/7/tasks/81f2d788-9d8b-4133-99ac-484b4a063873/orthophoto/**tiles/{zoom}/{x}/{y}.png**

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Just used the OSM Digitize plugin, and the tiles didnt work so I had to find the fix.

Thanks for this amazing plugin and project.

pierotofy, 2020-01-22:
Hey @JamesLeonDufour :hand: thanks for the report! Would you be interested in making a pull request for this? 

The relevant module is https://github.com/OpenDroneMap/WebODM/blob/master/plugins/osm-quickedit/public/main.js#L10

To setup a dev environment: https://docs.webodm.org/#docker-setup

JamesLeonDufour, 2020-01-23:
Yes please I found the solution @pierotofy

closed 2020-01-23 by smathermather