WebODM/WebODM #798: GDAL reported as missing or too old

dmbworld, 2020-01-22:
### What's your browser and operating system? (Copy/paste the output of 
Firefox 72 on WIndows 10
whatismybrowser.com/w/45KKWB6

### What is the problem?
Updated to lates version of WebODM and this problem reported:
===============================
Your system is currently using a version of GDAL that is too old, or GDAL is not installed. You need to install or configure your system to use GDAL 2.1 or higher. If you have installed multiple versions of GDAL, make sure the newer one takes priority in your PATH environment variable.
================================
Please see attached screen dump

### What should be the expected behavior?
GDAL is installed and is reported as:
GDAL 2.4.0, released 2018/12/14
from gdalinfo --version

WebODM Should start.
[image]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

WebODM no longer starts and is in a loop checking the GDAL version then reporting the error even though a newer version is installed

Please see screen dump as GDAL is reported as being installed
This installation has worked and been updated many times previously with no problem.

pierotofy, 2020-01-22:
Thanks for the report. Definitely a bug (fix coming soon).

pierotofy, 2020-01-22:
Should be fixed in #793. As soon as docker images are rebuilt, try to update and confirm this is resolved?

closed 2020-01-22 by pierotofy