WebODM/WebODM #803: WebODM/Docker Failing Due to Lack of Hard Drive Space, but Plenty is Allocated in Docker Desktop Settings and is Not Being Allocated

ludeykrus, 2020-01-30:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 79 on Windows 10

### What is the problem?

I'm running WebODM through Docker Desktop on Windows 10 Pro, running 2x Intel Xeon 2697V3's, 128gb registered RAM, 1TB SSD. WebODM runs successfully. It will complete processing projects. However, after a few projects, I am getting errors due to not able to write due to no disk space available. I noticed that in WebODM's diagnostic window, only ~58 gb of disk space and ~62gb of RAM are being utilized by WebODM/Docker despite having allocated 500gb of disk space and 100gb of RAM (allocated via Docker Desktop -> settings -> advanced). 

Last night before bed, I queued up 5 jobs to run over night. Woke up and the first two were still in their early stages, stuck since 2:30 AM. Docker ran out of disk space again at ~58.4gb and died. Before starting the jobs, I had tried the workaround in the following link involving changing the MobiLinux config file option for VHDsize, reset Docker settings to factory, and rebuilt containers for WebODM: (https://github.com/docker/for-win/issues/1042). It didn't work. Below is some Docker settings/readouts:

$ docker info
Client:
Debug Mode: false
Server:
Containers: 5
Running: 5
Paused: 0
Stopped: 0
Images: 39
Server Version: 19.03.5
Storage Driver: overlay2
Backing Filesystem: extfs
Supports d_type: true
Native Overlay Diff: true
Logging Driver: json-file
Cgroup Driver: cgroupfs
Plugins:
Volume: local
Network: bridge host ipvlan macvlan null overlay
Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
Swarm: inactive
Runtimes: runc
Default Runtime: runc
Init Binary: docker-init
containerd version: b34a5c8af56e510852c35414db4c1f4fa6172339
runc version: 3e425f80a8c931f88e6d94a8c831b9d5aa481657
init version: fec3683
Security Options:
seccomp
Profile: default
Kernel Version: 4.19.76-linuxkit
Operating System: Docker Desktop
OSType: linux
Architecture: x86_64
CPUs: 56
Total Memory: 61.52GiB
Name: docker-desktop
ID: 5CWM:BWTV:WWRT:2QED:QOI5:HTSO:5UTZ:GMWL:4GET:G6GH:CRTE:TOPV
Docker Root Dir: /var/lib/docker
Debug Mode: true
File Descriptors: 83
Goroutines: 115
System Time: 2020-01-29T21:44:57.650126882Z
EventsListeners: 4
Registry: https://index.docker.io/v1/
Labels:
Experimental: false
Insecure Registries:
127.0.0.0/8
Live Restore Enabled: false
Product License: Community Engine

### What should be the expected behavior?

I should be able to run WebODM through Docker successfully using the amount of disk space specified in the Docker Desktop GUI settings tab, instead of running out at ~60gb when I have almost a terabyte available. Although it hasn't actually stopped a process yet, if the under-allocated RAM ultimately causes issues, the full specified amount of RAM should also be utilized. 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

It happens every time when processing on a new build with new install of Docker/WebODM. Parameters are FastOrtho, no image resizing. Datasets have been sets of 450-750 JPEG images 12MP in size from a Mavic Pro.

[image]
[image]
[image]

pierotofy, 2020-01-30:
Mm, very strange, I don't know the exact cause, but seems like a problem with docker. WebODM does not limit the disk space in any way, so whatever is causing it is due to docker's odd behaviors on Windows.

bertramt, 2020-01-31:
I only have experience on Linux hosts but any of the times I've ran into space issues with webodm I've ran the ./webodm.sh down and then ./webodm.sh start. Not sure if that it's related to your same issue but it I've ran into it many times to the point where I'll rarely process a second large job without first running the down command.   I my very limited understanding on my issue i think one of the docker volumes is filling up and possibly leaving some temp data behind that is not being cleaned up after a run.  I've never sat down long enough to dig into it more.

vascodx, 2020-02-13:
Hi, same problem here, only 58.4 GB of storage available, not problem with ram (only 32GB on my PC),  docker 2.2.0.3 on Windows 10 x64

[image]

Client:
 Debug Mode: false

Server:
 Containers: 5
  Running: 5
  Paused: 0
  Stopped: 0
 Images: 4
 Server Version: 19.03.5
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: b34a5c8af56e510852c35414db4c1f4fa6172339
 runc version: 3e425f80a8c931f88e6d94a8c831b9d5aa481657
 init version: fec3683
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 4.19.76-linuxkit
 Operating System: Docker Desktop
 OSType: linux
 Architecture: x86_64
 CPUs: 4
 Total Memory: 23.49GiB
 Name: docker-desktop
 ID: H4JU:TZ5M:VYZJ:EOQW:ESMR:UVHT:JCXY:YBDZ:NE2N:YYE6:PZNO:53H6
 Docker Root Dir: /var/lib/docker
 Debug Mode: true
  File Descriptors: 80
  Goroutines: 112
  System Time: 2020-02-13T19:54:44.520477904Z
  EventsListeners: 4
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false
 Product License: Community Engine

ludeykrus, 2020-02-13:
So I believe I'm running into the issue listed in the link below:
https://github.com/docker/for-win/issues/1042?fbclid=IwAR192s6iUxH9HSiezWY7Zf_BrGAW57cLZ2fcZRO3Fx_tTRP-Mb9d9ARQYkU

Towards the bottom, a guy lists a link to his workaround which is basically what got the hard drive space issue fixed for me, but instead of editing just one entry in the MobyLinux.ps1 file for "VHDsize" variable setting, I had to find and replace all VHDsize variable declarations in the document for it to 'take'... I think there were 3-4 total. Update the VHDsize variables, restart Docker, reset all settings to factory defaults, and let WebODM rebuild and then my HD space was set to the MobyLinux settings, not what was entered in Docker Resources settings. I also changed the "Memory" variable declaration in MobyLinux.ps1 to a higher value, which did NOT fix/override the RAM allocation issue, but luckily for now I'm not RAM limited in my processing.

pierotofy, 2020-04-02:
Let's move this conversation over to https://community.opendronemap.org if the problem is still happening. :+1:

closed 2020-04-02 by pierotofy