WebODM/WebODM #807: WebODM.sh Start error starting worker & webapp, after adding user to docker Group

johnmbailey, 2020-02-05:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Your web browser is:
Firefox 72 on Ubuntu Linux

✓ Your web browser is up to date.

### What is the problem?

Receiving errors and unable to start WebODM due to errors, which contine after adding user to Docker group.
john@john-Satellite-L855:~/WebODM$ ./webodm.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1
Starting db       ... 
Starting node-odm ... 
Starting broker   ... 
Starting worker   ... error
Starting webapp   ... error

ERROR: for db  ('Connection aborted.', error(13, 'Permission denied'))

ERROR: for node-odm  ('Connection aborted.', error(13, 'Permission denied'))

ERROR: for broker  ('Connection aborted.', error(13, 'Permission denied'))
ERROR: Couldn't connect to Docker daemon at http+docker://localunixsocket - is it running?

If it's at a non-standard location, specify the URL with the DOCKER_HOST environment variable.
ERROR: Couldn't connect to Docker daemon at http+docker://localunixsocket - is it running?

If it's at a non-standard location, specify the URL with the DOCKER_HOST environment variable.
john@john-Satellite-L855:~/WebODM$ sudo adduser john docker
[sudo] password for john: 
The user `john' is already a member of `docker'.

### What should be the expected behavior?

Error is preventing Docker / WebODM from starting

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I followed instructions, using Ubuntu 18 latest updated version.

pierotofy, 2020-02-05:
Hey @johnmbailey :hand: try rebooting your computer or log-in and out.

johnmbailey, 2020-02-05:
Thanks for the reply.  I logged out and then logged back in before posting the original errors of the message.  I shut down Ununtu and restarted, and itthen worked. I guess it required a full restart.

closed 2020-02-05 by johnmbailey

pierotofy, 2020-02-06:
:+1:  would you be interested in updating the docs? Perhaps this should be added. https://github.com/OpenDroneMap/docs/#opendronemap-docs