WebODM/WebODM #810: [Feature Request] Allow admin to instantly identify which user created the specific project

yeenie, 2020-02-10:
It will be great if the admin can monitor all the projects created by every other user. For example, when a user creates a project, the owner/creator of the project is shown together with the project at the dashboard. It will be better if admin can arrange/categorize projects into dashboard folders based on owner/creator to easily organize and manage all projects.

pierotofy, 2020-02-10:
Hi @yeenie,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-02-13 00:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2021-11-01:
With the new permission management box (pressing `Edit` on a project from the dashboard) an admin can quickly see who owns the project, so this should be implemented :pray:

closed 2021-11-01 by pierotofy