WebODM/WebODM #811: Stuck in Post-Processing Step

cgiovando, 2020-02-11:
### What's your browser and operating system?

whatismybrowser.com/w/297BH8D

### What is the problem?

WebODM gets stuck during post-processing step.

### What should be the expected behavior?

The process should finish correctly.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Fresh WebODM setup on Windows, following the latest installation guide.

[image]

Log is available here https://www.dropbox.com/s/699k3sxuwdgqdr6/WebODM%20Console%20Log%202020-02-11.txt?dl=0

Many thanks for your help!

smathermather, 2020-02-11:
I hate to burn CPU cycles, but can you cancel and restart? I am curious if this is repeatable on your machine.

pierotofy, 2020-07-27:
Please see if #893 solves this issue by updating and retrying. If the problem is still there, please reopen? :pray:

closed 2020-07-27 by pierotofy