WebODM/WebODM #819: ./webodm.sh start --dev doesn't work

yeenie, 2020-02-19:
### What's your browser and operating system? 
Google chrome
Microsoft Windows 10 Pro

### What is the problem?

I want to edit the sourcecode but ./webodm.sh start --dev doesn't work. I tried following the steps as below. 
1. Make a fork of the WebODM repository
2. Clone your repository in a directory
3. Create a new branch: git checkout -b branchname.
4. Setup a development environment either with docker or natively.

I'm stuck at ./webodm.sh start --dev because it just doesn't seem to load/open? It just loads a bunch of stuff for a long time and seem to be stuck and I still can't open webodm. 

### What should be the expected behavior?

Being able to open webodm in localhost and make changes in sourcecode that is reflected in the running WebODM instance automatically.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

CL-GOHYN+office@CL-GohYN MINGW64 ~
$ git clone https://github.com/yeenie/WebODM --config core.autocrlf=input --depth 1
Cloning into 'WebODM'...
remote: Enumerating objects: 1097, done.
remote: Counting objects: 100% (1097/1097), done.
remote: Compressing objects: 100% (947/947), done.
remote: Total 1097 (delta 148), reused 705 (delta 100), pack-reused 0
Receiving objects: 100% (1097/1097), 31.73 MiB | 1.82 MiB/s, done.
Resolving deltas: 100% (148/148), done.
Updating files: 100% (1004/1004), done.

CL-GOHYN+office@CL-GohYN MINGW64 ~
$ cd WebODM

CL-GOHYN+office@CL-GohYN MINGW64 ~/WebODM (master)
$ ./webodm.sh start --dev
Checking for docker...   OK
Checking for git...   OK
Checking for docker-compose...   OK
Starting WebODM in development mode...
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.nodemicmac.yml down --remove-orphans
Removing network webodm_default
Network webodm_default not found.

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key:
SSL certificate:
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.dev.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.dev.yml up --scale node-odm=1
Starting node-odm ... failed
Starting broker   ... failed
Starting db       ... failed
Starting worker   ... failed
Starting webapp   ... failed
No containers to start
Creating network "webodm_default" with the default driver
Creating broker            ... done
Creating webodm_node-odm_1 ... done
Creating db                ... done
Creating worker            ... done
[... 689 lines cut ...]
webapp      |     [./node_modules/css-loader/index.js!./node_modules/sass-loader/dist/cjs.js?!./app/static/app/js/css/UploadProgressBar.scss] ./node_modules/css-loader!./node_modules/sass-loader/dist/cjs.js??ref--5-2!./app/static/app/js/css/UploadProgressBar.scss 228 bytes {0} [built]
webapp      |         + 1 hidden module
webapp      | Child extract-text-webpack-plugin node_modules/extract-text-webpack-plugin/dist node_modules/css-loader/index.js!node_modules/sass-loader/dist/cjs.js??ref--5-2!app/static/app/js/vendor/google-code-prettify/prettify.css:
webapp      |     Entrypoint undefined = extract-text-webpack-plugin-output-filename
webapp      |     [./node_modules/css-loader/index.js!./node_modules/sass-loader/dist/cjs.js?!./app/static/app/js/vendor/google-code-prettify/prettify.css] ./node_modules/css-loader!./node_modules/sass-loader/dist/cjs.js??ref--5-2!./app/static/app/js/vendor/google-code-prettify/prettify.css 1.31 KiB {0} [built]
webapp      |         + 1 hidden module
webapp      | Child extract-text-webpack-plugin node_modules/extract-text-webpack-plugin/dist node_modules/css-loader/index.js!node_modules/sass-loader/dist/cjs.js??ref--5-2!app/static/app/js/vendor/leaflet/L.Control.MousePosition.css:
webapp      |     Entrypoint undefined = extract-text-webpack-plugin-output-filename
webapp      |     [./node_modules/css-loader/index.js!./node_modules/sass-loader/dist/cjs.js?!./app/static/app/js/vendor/leaflet/L.Control.MousePosition.css] ./node_modules/css-loader!./node_modules/sass-loader/dist/cjs.js??ref--5-2!./app/static/app/js/vendor/leaflet/L.Control.MousePosition.css 416 bytes {0} [built]
webapp      |         + 1 hidden module
webapp      | Child extract-text-webpack-plugin node_modules/extract-text-webpack-plugin/dist node_modules/css-loader/index.js!node_modules/sass-loader/dist/cjs.js??ref--5-2!app/static/app/js/vendor/leaflet/Leaflet.Autolayers/css/leaflet.auto-layers.css:
webapp      |     Entrypoint undefined = extract-text-webpack-plugin-output-filename
webapp      |     [./app/static/app/js/vendor/leaflet/Leaflet.Autolayers/images/arrow-down.png] 489 bytes {0} [built]
webapp      |     [./app/static/app/js/vendor/leaflet/Leaflet.Autolayers/images/arrow-up.png] 481 bytes {0} [built]
webapp      |     [./app/static/app/js/vendor/leaflet/Leaflet.Autolayers/images/close.png] 313 bytes {0} [built]
webapp      |     [./app/static/app/js/vendor/leaflet/Leaflet.Autolayers/images/remove.png] 241 bytes {0} [built]
webapp      |     [./node_modules/css-loader/index.js!./node_modules/sass-loader/dist/cjs.js?!./app/static/app/js/vendor/leaflet/Leaflet.Autolayers/css/leaflet.auto-layers.css] ./node_modules/css-loader!./node_modules/sass-loader/dist/cjs.js??ref--5-2!./app/static/app/js/vendor/leaflet/Leaflet.Autolayers/css/leaflet.auto-layers.css 3.48 KiB {0} [built]
webapp      |         + 1 hidden module
webapp      | Child extract-text-webpack-plugin node_modules/extract-text-webpack-plugin/dist node_modules/css-loader/index.js!node_modules/sass-loader/dist/cjs.js??ref--5-2!app/static/app/js/vendor/potree/libs/spectrum.css:
webapp      |     Entrypoint undefined = extract-text-webpack-plugin-output-filename
webapp      |     [./node_modules/css-loader/index.js!./node_modules/sass-loader/dist/cjs.js?!./app/static/app/js/vendor/potree/libs/spectrum.css] ./node_modules/css-loader!./node_modules/sass-loader/dist/cjs.js??ref--5-2!./app/static/app/js/vendor/potree/libs/spectrum.css 15 KiB {0} [built]
webapp      |         + 1 hidden module
webapp      | Child extract-text-webpack-plugin node_modules/extract-text-webpack-plugin/dist node_modules/css-loader/index.js!node_modules/sass-loader/dist/cjs.js??ref--5-2!node_modules/leaflet/dist/leaflet.css:
webapp      |     Entrypoint undefined = extract-text-webpack-plugin-output-filename
webapp      |        5 modules

pierotofy, 2020-02-19:
Problem is here:

```
webapp | fatal: bad object d6240e0bfcfed59faf416a3d67c2d5aa902ec950
webapp | fatal: remote did not send all necessary objects
```

Either a network issue, or Windows is having fusses with the volume mount.

pierotofy, 2020-02-19:
Tested on Windows 10 Pro also, but was able to get it running:

[image]

Check that some program in the background is not blocking your network connections or interfering with the volume mounts (is the C:\ drive shared within docker?)

[image]

yeenie, 2020-02-19:
@pierotofy Yes the C:\ drive is shared within docker already but it still doesn't work :(  How should I solve this?

pierotofy, 2020-02-19:
I'm really out of ideas :( I don't know what's causing it.

Turboleb, 2020-06-19:
I'm having the same issue here though I'm not running in development mode.
./webodm.sh start

Using the following environment:
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key:
SSL certificate:
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.dev.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.dev.yml up --scale node-odm=1
Starting node-odm ... failed
Starting broker ... failed
Starting db ... failed
Starting worker ... failed
Starting webapp ... failed
No containers to start

CarlosHocker, 2020-07-13:
Do you have enough free space on your drive?
Freeing some space fixed it for me

pvgmenegasso, 2021-05-21:
I have the same problem with rio-tiler, but I am trying to run --dev on a corporate firewall.
I have managed to workaround for all the other requirements, but this one is a little pesky.
Does anyone knows if there is a way I could download it elsewhere and transfer it directly via rsync or something similar ? @pierotofy if you have any insights on this it would be awesome :)
I also did not quite well understood why this specific requirement is so different from the others, is this because you need a developer version of the module ? Is there a way to do it besides using the git:// link ? Like specifying module name and them adding --dev tag or something like that ?
rio_tiler --dev for instance.