WebODM/WebODM #823: problem with WebODM node-odm-1 seems to be offline

cupax64, 2020-02-24:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[whatismybrowser.com/w/EHYMM3S]

### What is the problem?

[I have DockerToolbox-19.03.1 on windows 10 x64. He cloned the WebODM repository, saved it in D: \ and ran ./webodm.sh start Now, I have created an administrator user, it has all permissions but the node does not work. I have read more comments about this problem, but it has not been resolved by restarting docker or reinstalling.

I receive this -> node-odm-1 seems to be offline. I have also created another node, in port 8000 and it doesn't work for me, it always puts the same thing on me, offline node. I need help.]

### What should be the expected behavior?

[That the node is not offline]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[I still couldn't load data on the page, I just installed the docker toolbox, found the command, and the node error already appears. 
[image]
]

Pilskalns, 2020-02-25:
Having same issue.

Since this is is my first time using ODM, poked around how that all works.
At some point figured out that default hostname inside ODM config is the container name and probably in the internal network that has issues to resolve.

1. With the following command got internal IP:
[image]

2. Updated ODM config:
[image]

3. Profit.

pierotofy, 2020-02-25:
I wonder if something has changed in the latest docker releases (this seems to be an issue with our automated testing suite, which started failing a few weeks ago).

Mind asking what version of docker you are running?

Pilskalns, 2020-02-25:
@pierotofy 

Docker version 19.03.5, build 633a0ea
Got it from Chocolatey `docker-desktop` package.

ODM downloaded with the following command:
`git clone https://github.com/OpenDroneMap/WebODM.git --recurse-submodules .`

pierotofy, 2020-02-25:
Windows Pro, also docker 19.03.5:

[image]

Strange.

What's the output of:

```
docker exec -ti webapp cat /etc/resolv.conf
```

I get: 

```
# docker exec -ti webapp cat /etc/resolv.conf
nameserver 127.0.0.11
options ndots:0
```

Also:

```
# docker exec -ti webapp bash
root@16421257525a:/etc# apt install -y dnsutils
root@16421257525a:/etc# dig webodm_node-odm_1

; <<>> DiG 9.11.5-P4-5.1-Debian <<>> broker
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 38670
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;broker.                                IN      A

;; ANSWER SECTION:
broker.                 600     IN      A       172.19.0.2

;; Query time: 0 msec
;; SERVER: 127.0.0.11#53(127.0.0.11)
;; WHEN: Tue Feb 25 23:37:55 UTC 2020
;; MSG SIZE  rcvd: 46
```

cupax64, 2020-02-26:
> Having same issue.
>
> Since this is is my first time using ODM, poked around how that all works.
> At some point figured out that default hostname inside ODM config is the container name and probably in [...]

> Having same issue.
>
> Since this is is my first time using ODM, poked around how that all works.
> At some point figured out that default hostname inside ODM config is the container name and probably in [...]

[image]

He put the same command as you, and it doesn't work for me, he tried the node name as a parameter, and neither, what are you doing wrong?

cupax64, 2020-02-26:
Already solved. I put my answer in case someone is worth it. Use docker ps to see the id of the container. Taking the id, which is the first column, docker inspects 'id' and the ip appears to me, then, indeed, as @Pilskalns says, I put the name, the ip and it worked. Thank you all very much, and sorry if my English is bad, I am Spanish.

pierotofy, 2020-02-26:
I'll keep this open since this is a workaround, but not a fix.

swarmdog, 2020-04-07:
This is still an issue with the current build, you must docker network inspect webodm_default to find the node-odm1 IP and update it via the dashboard.

ibrahimcengiz, 2020-07-07:
I have tried everything mentioned on this page, nothing solves my problem.

smathermather, 2020-07-08:
@ibrahimcengiz -- come on over to https://community.opendronemap.org and let's see if we can help you troubleshoot.

AnthonyCoughlin, 2021-10-12:
@guimandola 

I'm not sure if this is ridiculous to throw out there, but I'm doing it....  So...  just looking at your screenshot and comparing to my setting, your 'Hostname' value has underscores and dashes  Mine has only dashes.  Wouldn't it be beautiful if it was that simple??

[image]

guimandola, 2021-10-13:
Perfect, it worked
thx

MiCkaelN72, 2022-08-12:
[image]
o nome da pasta que esta os arquivos do programa tem que ser igual ao nome do host

KwahuNashoba, 2022-08-15:
It seems to be the problem that if you download zip of the repo (instead of cloning it) you will end up with folder WebODM-master from where you will build and lunch docker files. Because of that, your node container will be named webodm-master_node-odm_1 and the default hostname is set to webodm_node-odm_1. So replace the webodm part with however your folder is named :) Hope this helps