WebODM/WebODM #826: Upgrade to PostgreSQL 12.x

Saijin-Naib, 2020-02-27:
Similar to the move to Python3 and possibly to latest LTS (20.04), PostgreSQL 9.x is deprecated above 16.04 LTS and is replaced by PostgreSQL 10.x on 18.04.3 and 12.x on 20.04.

18.04.3:  
https://packages.ubuntu.com/bionic/database/postgresql

20.04:  
https://packages.ubuntu.com/focal/database/postgresql

pierotofy, 2020-02-27:
Hi @Saijin-Naib,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-03-01 15:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Saijin-Naib, 2020-02-27:
Thanks again, Piero. I still don't understand what dependencies are for each component only 🤣

Of course, we might be able to avoid the whole PGSQL upgrade dance with OGC GeoPackage ;)
https://community.opendronemap.org/t/ogc-geopackage-for-data-storage-back-end/3650?u=saijin_naib

ssantacroce-uwach, 2022-10-11:
@pierotofy I figured I would take a stab at this, since 9 is SUPER old... any reason we are using our own dockerfile vs using a base postgis image like postgis/postgis:14-3.3 and throwing in scripts in the init.db folders?? 

Switching to a base image should make it a lot easier to upgrade versions

pierotofy, 2022-10-12:
We use our own (compiled from source) version of postgres/postgis, because they don't publish 9.x releases on docker hub anymore.

The problem with postgres/postgis, using docker, is that there's no easy way to perform a fully automated migration for dozens of thousands of users that are using WebODM via docker.

The correct way to perform the migration, is to create a new docker image which has the newer postgres/postgis version and perform the migration on startup. Postgres does not offer an automated way to do this, so it's going to be a bit tricky.

In the meanwhile, Postgres 9.x continues to work just fine (aside from being old, we haven't had a *good* reason to migrate).

ssantacroce-uwach, 2022-10-12:
That makes sense. I will start working on the migration scripts first then

ssantacroce-uwach, 2022-10-13:
https://www.cloudytuts.com/tutorials/docker/how-to-upgrade-postgresql-in-docker-and-kubernetes/

Seems pretty straight forward. So I'm thinking we add a upgrade_db option to webodm.sh. That will do a docker compose up to a new file docker-compose.upgrade_db.yml. This will stand up the new database. Then it will run the pg_dumpall on the old database. Then we will push the dump file into the new databse. Finally we will stop the old one. Since the base docker-compose would already have been "updated" it will work correctly going forward. 

I am sure I'm missing some steps here. But that's where I will be starting

smathermather, 2022-10-13:
Sounds great. I've got a server with lots of projects in it and the ability to snapshot it at the VM level, so I'd be happy to be a ~~victim~~ volunteer for testing.

pierotofy, 2022-10-13:
The problem with the proposed approach is that it's not automated; it requires user intervention. Please bear in mind that for us technical folks using the command line this is not a problem, but there are lots of folks that have installed WebODM via the automated installers, and this approach is going to break their installation and does not provide a path for them to upgrade.

ssantacroce-uwach, 2022-10-13:
Who said anything about manual? Everything in that article I believe can be
automated. If not, then maybe it's time to have a 2nd offering.

I agree that many are not tech savvy, but having vulnerable software
running on their computers can't be good either.

ssantacroce-uwach, 2022-10-14:
Ok so I'm moving over to the pull request I'm about to make. It's nowhere NEAR ready, but it's a start, and since I'm still very green with the codebase I will still need some hand holding! But running the commands by hand dumped and pushed the data from 9.6 to 14!

NtskwK, 2024-09-08:
The version of PostgreSQL was been writen in `$PGDATA/pg_hba.conf`, we can use it to determine if we need to upgrade. Then, [pgupgrade](https://www.postgresql.org/docs/current/pgupgrade.html) can help us with data volume migration. 

> 我们使用我们自己的（从源代码编译的）版本的 postgres/postgis，因为它们不再在 docker hub 上发布 9.x 版本。
> 
> 使用 docker 的 postgres/postgis 的问题在于，没有简单的方法可以为成千上万通过 docker 使用 WebODM 的用户执行完全自动化的迁移。
> 
> 执行迁移的（可能）正确方法是创建一个具有较新 postgres/postgis 版本的新 docker 镜像，并在启动时执行迁移。该过程很混乱，因为 postgres 没有提供自动化方法来执行此操作。所以这会有点棘手。
> 
> 与此同时，Postgres 9.x 继续正常工作（除了旧之外，我们还没有_充分_的理由进行迁移）。
> 
> 我们欢迎帮助执行此操作，但请注意，仅更新到 12.x _是不够_的，还需要从 9.x --> 12.x 迁移现有数据卷，并且需要完全自动化（适用于所有平台）。`./webodm.sh update`

NtskwK, 2024-09-08:
> We use our own (compiled from source) version of postgres/postgis, because they don't publish 9.x releases on docker hub anymore.
> 
> The problem with postgres/postgis, using docker, is that there's no easy way to perform a fully automated migration for dozens of thousands of users that are using WebODM via docker.
> 
> The (probably) correct way to perform the migration, is to create a new docker image which has the newer postgres/postgis version and perform the migration on startup. The procedure is messy, as postgres does not offer an automated way to do this. So it's going to be a bit tricky.
> 
> In the meanwhile, Postgres 9.x continues to work just fine (aside from being old, we haven't had a _good_ reason to migrate).
> 
> We'd welcome help to do this, but note that simply updating to 12.x is _not_ sufficient, the migration of existing data volumes from 9.x --> 12.x is also needed, and needs to be fully automated (work via `./webodm.sh update` for all platforms).

The version of PostgreSQL was been writen in `$PGDATA/pg_hba.conf`, we can use it to determine if we need to upgrade. Then, [pg_upgrade](https://www.postgresql.org/docs/current/pgupgrade.html) can help us with data volume migration. There is a problem  with `live update` functions. `pg_upgrade` can only be used when both database servers are stopped using. New servers can be stopped, but old cannot while webodm is running.

pierotofy, 2025-03-12:
We will upgrade when there will be a clear benefit or reason (e.g. needing to upgrade Django), or if someone sponsors it.

closed 2025-03-12 by pierotofy

pierotofy, 2026-10-30:
This is now done in #2044 🥳