WebODM/WebODM #833: Volume Measurement reports "Invalid task output"

dmbworld, 2020-03-13:
### What's your browser and operating system?
whatismybrowser.com/w/ADN7Z5H

### What is the problem?

Volume measurement reports  "Invalid task output (cannot find valid key)"

### What should be the expected behavior?

Should report volume of area defined
[image]

### How can we reproduce this? 
Attempt a volume measurement

pierotofy, 2020-03-13:
Confirm, introduced by mistake in #829

pierotofy, 2020-03-13:
Wait until docker images are rebuilt (~1 hour), update and test if it's fixed?

Thanks for the report @dmbworld ! :pray:

PR #834 (2020-03-13): Fix #833
Fixes #833
commit c6234414 (2020-03-13): Merge pull request #834 from pierotofy/grassimp

Fix #833
commit cb6fbca5 (2020-03-13): Fix #833
closed 2020-03-13 by pierotofy