WebODM/WebODM #837: Plant Health Layer LAI export GeoTiff wrong dtype expects float64 got float32

woodbumper, 2020-03-21:
Hi,

when trying to export an LAI Layer in Plant Health, I get the following error:

_the array's dtype 'float64' does not match the file's dtype 'float32'_

I wonder if it's possible to implement either a exception routine 

_exception TypeError -> type cast float64 -> float32_

or (more likely)
 
if the in the original layer stored value exceeds the upper limit of float32 transforming the stored values into comparable values of the GeoTiff.

[image]

As the Plant Health Layers are of some importants for agriculture and forestry, a solution/fixing of the error would be nice.

Thanks in advance.

pierotofy, 2020-03-21:
Hi @woodbumper,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-03-24 09:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2020-03-21:
This should be fixed in #836 :+1: 

Thanks for the report!

Wait a few hours until the docker images rebuild, then update and try again. Re-open this issue if the problem persists.

commit a874b80f (2020-03-21): Fix #837
closed 2020-03-21 by pierotofy