WebODM/WebODM #844: Export - GRASS script error

cj-cuccaro, 2020-03-28:
### What's your browser and operating system? 
whatismybrowser.com/w/CLAKZ4MType answer here]

### What is the problem?

When I try to do an export from the Elevation Map menu I get the following error message...
Could not execute GRASS script /webodm/plugins/elevationmap/elevationmap.py from /webodm/app/media/tmp/tmpbjr8at8b_grass_engine: Starting GRASS GIS... Creating new GRASS GIS location <location>... Cleaning up temporary files... Executing <python /webodm/plugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/9/task/4d9ef9a3-00a9-4137-89f5-df02236c0249/assets/odm_dem/dsm.tif interval=5 format=ESRI Shapefile noise_filter_size=3.0 epsg=4326> ... Traceback (most recent call last): File "/webodm/plugins/elevationmap/elevationmap.py", line 48, in <module> import cv2, math, argparse ModuleNotFoundError: No module named 'cv2' Execution of <python /webodm/plugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/9/task/4d9ef9a3-00a9-4137-89f5-df02236c0249/assets/odm_dem/dsm.tif interval=5 format=ESRI Shapefile noise_filter_size=3.0 epsg=4326> finished. Cleaning up temporary files...

### What should be the expected behavior?

An exported file instead of an error message. see attached image
[image]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

From the WebODM Dashboard click on the View Map for a completed task to display the orthophoto. Select anyone of the export options shown in the Export menu button and get the error message shown above.

I did an update from the WebODM Manager and noticed some GRASS script issues in the log. I attached a file with this information.
[WebODM-log.txt](https://github.com/OpenDroneMap/WebODM/files/4397763/WebODM-log.txt)

Please let me know if you need further information on this issue.
Thanks,
CJ

pierotofy, 2020-03-29:
Should be fixed with #843 :+1: 

If this is still an issue after updating, please re-open it. Thanks!

closed 2020-03-29 by pierotofy

cj-cuccaro, 2020-03-29:
Happy to report that fixed the problem.
Thank you,
CJ