WebODM/WebODM #845: ERROR: Couldn't connect to Docker daemon at http+docker://localhost - is it running?

leodaft, 2020-03-31:
Hi, I'm trying to install WebODM on my desktop PC, but when I run "./webodm.sh start" I get the following error. As per instructions, I first installed Docker. I also tried to install NodeODM which works properly instead. Can you help me?

On my pc I have: Ubuntu 18.04.4 LTS;
Intel® Core™ i5-4440 CPU @ 3.10GHz × 4
93,3 GB Hard Disk
7,5GiB DDR3 RAM

~/WebODM$ ./webodm.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1
ERROR: Couldn't connect to Docker daemon at http+docker://localhost - is it running?

If it's at a non-standard location, specify the URL with the DOCKER_HOST environment variable.
ERROR: Couldn't connect to Docker daemon at http+docker://localhost - is it running?

If it's at a non-standard location, specify the URL with the DOCKER_HOST environment variable.

pierotofy, 2020-03-31:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

closed 2020-03-31 by pierotofy

leodaft, 2020-03-31:
ok