WebODM/WebODM #855: No visible processing node for non-superuser

supersxx, 2020-04-25:
Everything works well for the superuser account, but there is no visible processing node for other types of account unless I assign it the superuser privilege. Can anyone help?

pierotofy, 2020-04-25:
Hi @supersxx,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-04-28 04:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

supersxx, 2020-04-25:
Thank you. Here is some more information:
What's your browser and operating system? 
Chrome 81 on Windows 10

What is the problem?
Nonsuperuser doesn't have any visible processing node, and cannot start any task. Everything works well for the superuser account

What should be the expected behavior?
The processing node node-odm-1 appears in the sidebar list and available to select when start a new task 

How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
I used docker on windows10 following the instructions from github to install webodm without making any change

supersxx, 2020-04-25:
by the way, the guest account has all authorizations except for superuser

supersxx, 2020-04-25:
Problem solved. Please delete the issue. Sorry for the inconvenience.

smathermather, 2020-04-25:
What was the issue @supersxx?

closed 2020-04-25 by smathermather