WebODM/WebODM #862: Error: No containers to start

robbiebonds, 2020-05-11:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Firefox 76 on Windows 10 Pro]

### What is the problem?

[After entering the command, "./webodm.sh start" I get the output below. 

Checking for docker...   OK
Checking for git...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key:
SSL certificate:
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1
Starting node-odm ... failed
Starting broker   ... failed
Starting db       ... failed
Starting worker   ... failed
Starting webapp   ... failed
No containers to start
Pulling node-odm (opendronemap/nodeodm:)...
latest: Pulling from opendronemap/nodeodm]

### What should be the expected behavior?

[I expect WebODM to be accessible in my browser and for a container to be visible on Docker Desktop]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[ Unfortunately, I'm not sure. My Docker Desktop settings are below and I've copied the GIT Bash output. If any other information would be useful, please let me know.
Resources
Advanced
CPUs: 5

Memory: 4.75 GB

Swap: 3 GB

Disk image size: 123 GB (6.3 GB used) ]

smathermather, 2020-05-12:
Let's move this over to https://community.opendronemap.org. If it's confirmed as a bug, we can re-open this, but it looks like a support issue.

closed 2020-05-12 by smathermather

Turboleb, 2020-06-18:
I'm having the exact problem too. Any solution yet please?

robbiebonds, 2020-06-18:
So far no joy. Had to run it through command prompt.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/OpenDroneMap/WebODM/issues/862#issuecomment-646345101",
"url": "https://github.com/OpenDroneMap/WebODM/issues/862#issuecomment-646345101",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]