WebODM/WebODM #864: Problem getting WebODM to run properly (was previously working)

dronecloudtech, 2020-05-20:
Hello, i'm facing a problem since few days when i have ugrade my system linux to ubuntu 20.04LTS and i have installed webodm but when i want to upload my file i get this errow bellow:
```There are no usable processing nodes. Make sure that at least one processing node is reachable and that you have granted the current user sufficient permissions to view the processing node (by going to Administration -- Processing Nodes -- Select Node -- Object Permissions -- Add User/Group and check CAN VIEW PROCESSING NODE). If you are bringing a node back online, it will take about 30 seconds for WebODM to recognize it.```
please help me to resolve it

i have add all right to my users on webodm but is the same problem, i have reinstalled my computer but it's the same

pierotofy, 2020-05-20:
Hi @dronecloudtech,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-05-23 08:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

dronecloudtech, 2020-05-20:
OK i will try it

closed 2020-05-20 by pierotofy