WebODM/WebODM #865: View 3D Model fails to load since #863 merged to master.

dmbworld, 2020-05-22:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

whatismybrowser.com/w/L6EMD38

### What is the problem?

After updating WebODM to version with #863 "View 3D Model" fails to load mesh and no error messages displayed.

### What should be the expected behavior?

Mesh view should be displayed

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Updated after #863 was merged in to Master. ALL existing tasks fail to load 3D Mesh.

[image]

After selecting "View 3D Model:
[image]

View Map (2D) does work:
[image]

Waited for over 3 hours before updating to ensure that Docker had been uptated.
In an effort to resolve probelm used the rebuild option of "webodm.sh" and problem still exists.

timotay07, 2020-05-22:
I can confirm that this is an issue that we are experiencing using both Firefox and Chrome.  We ran the update script yesterday evening and this morning.

pierotofy, 2020-05-22:
Thanks for the report! This was a bug indeed. Should be fixed (docker images will get rebuilt soon).

PR #867 (2020-05-22): Added potree
Fixes #865 

Beware of `.gitignore` patterns.
closed 2020-05-22 by pierotofy