WebODM/WebODM #866: Login fails for user create via API (password not processed or lacking documentation)

pqvst, 2020-05-22:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 81 on macOS (Catalina)

### What is the problem?

When creating a user via the API, the password does not work. It seems the API is not correctly processing the password, or the documentation doesn't state in which format the password should be provided.

### What should be the expected behavior?

I expect the login credentials to work.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

First generate an API auth token using your admin credentials:

```
curl -X POST -d "username=adminuser&password=adminpass" http://localhost:8000/api/token-auth/
```

Then invoke the "create a user" API endpoint:

```
curl -X POST -H "Authorization: JWT <token>" -d "username=testuser&password=testpass" http://localhost:8000/api/admin/users/
```

Here is the response. Note that the password is returned in cleartext:

```
{"id":10,"password":"testpass","last_login":null,"is_superuser":false,"username":"testuser","first_name":"","last_name":"","email":"","is_staff":false,"is_active":false,"date_joined":"2020-05-22T09:40:14.285065Z","groups":[],"user_permissions":[]}
```

When checking WebODM it shows that the password is in an invalid format:

[image]

I'm guessing the API is not processing the password (i.e. it is assuming it is in whatever internal format webodm uses to stored the salted/hashed password).

Is this a bug or a feature?

The API docs don't state anything about the format. Just says "password".

https://docs.webodm.org/#create-a-user

pierotofy, 2020-05-22:
Hey @pqvst :hand: I think the API expects the output of `make_password` as from:

```
>>> from django.contrib.auth.hashers import make_password
>>> make_password('test123')
'pbkdf2_sha256$120000$TACc1XZLIuwM$cwDupsXeMNMPL/TKs1OCRKpSlY7OkNaGNrW9Lrx7Pb8='
```

(Careful to escape the $ characters with \$ if you type the string into bash or some other terminal).

This is less than ideal, perhaps you might be interested in modifying the API behavior to accept a clear password instead? :pray: 

Or simply a note in the docs would be helpful to clarify this also.

abrbhat, 2020-10-21:
Sent https://github.com/OpenDroneMap/WebODM/pull/926

commit 75062afe (2020-10-23): Fix #866 - Hash password while creating user through admin API
commit d09f35b7 (2020-10-23): Merge pull request #926 from abrbhat/master

Fix #866 - Hash password while creating user through admin API
closed 2020-10-23 by pierotofy