WebODM/WebODM #869: Improved resume support

dawonn-haval, 2020-05-23:
### How did you install OpenDroneMap? (Docker, natively, ...)?

git clone <snip>
./webodm.sh start

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome, Ubuntu

### What is the problem?

I ran out of disk space, and want to resume where it left off after making space. 

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

The ability to 'restart from' is great but I don't know what steps have been completed and where to restart from. I wish there was some way to find out, or for it to figure it out automatically. 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Process a large dataset on a small hard disk.

KommandorKeen, 2020-07-06:
I have just run a few test cases through meshroom.  ODM produces a better result but has a poorer re-start visibility.
I think it comes down to re-structuring the options list so that they are delivered chronologically. I will have alook ate ths.

smathermather, 2020-07-06:
Love it. And I think to the question above from @dawonn-haval, ODM does it's own check as to what steps still need run. You only need to specify the restart stage if you've changed parameters that would effect a stage that has already run.

smathermather, 2020-07-06:
But I love restructuring the options list. It's a long and confusing list, and it would be quite helpful to add structure to it.

pierotofy, 2020-07-27:
Please see if #893 solves this issue by updating and retrying. If the problem is still there, please reopen? :pray:

closed 2020-07-27 by pierotofy