WebODM/WebODM #870: Memory & Storage historical plot

dawonn-haval, 2020-05-23:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Ubuntu, Chrome

### What is the problem?

When a task fails, It would be nice to be able to see a plot of the memory and storage use during the task to see if the system simply ran out of memory / storage.

### What should be the expected behavior?

The diagnostics tab shows a pie graph, which is great for seeing the current state of the system. It would be super helpful if this were a plot / graph so that we could look at how much RAM / Storage was in use during the processing. If it maxes out and then drops, it can be easily determined that more RAM is required. 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

process a large dataset with little ram, and let it run over night. Wake up disappointed and try again with more RAM... :)

dawonn-haval, 2020-05-23:
For anyone else in a similar situation; I installed Prometheus with the node_exporter 'plugin' to record  system metrics. That should solve the immediate need, but would still be nice to have some integrated solution. :)

pierotofy, 2020-05-23:
Indeed, I think this has been asked in the past too. It would be a matter of adding a prometheus docker-compose file that pulls the prometheus docker image and links it with the rest of the stack. Perhaps you might be interested in contributing to this?

dawonn-haval, 2020-05-23:
Would really love to, but my todo list is already long and full of terrors... :(

If you're gonna bundle it, Grafana would probably be a better user experience. The default prometheus installation is rather bare-bones and terse. (Intentionally so, I think)  

It might also be wise to investigate InfluxDB. I have heard good things, but ultimately I don't know much about the TSDB's, I just googled things until it did what I wanted... :)