WebODM/WebODM #871: Imported Task missing Camera Images

dmbworld, 2020-05-26:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

whatismybrowser.com/w/CB78MSW

### What is the problem?

Exporting/Importing of task does not copy Camera images as used by the new camera view.

### What should be the expected behavior?

Export a task using all assets from one system and then Import to another. The imported task should have the same data and allow viewing of Camera images.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters

Perform the previous explained expected behaviour.

[image]

Saijin-Naib, 2021-09-07:
Received customer report on Lightning as of today. Looks like still current.

pierotofy, 2021-09-07:
Yep; the `all.zip` does not contain the images, so they are missing when imported.

At some point we'll probably need a more formal "import/export" functionality in WebODM, that exports all info from a dataset, not just the output assets.

pierotofy, 2021-10-16:
After uploading the images, you would also need to create the appropriate records in the database (`app_image_upload` table). (I'd recommend writing a script for that).

markFieldman, 2021-11-14:
Recently I wrote script for this needs - you can use my [gist] for this purpose (https://gist.github.com/markFieldman/f2f0667d623848ba3561a1c4a4776237)

markFieldman, 2021-11-14:
```
### 1 copy this part with function end and paste to the manage.py shell
from app.models import ImageUpload, Task
from json import load as parse_json

def reindex_shots(projectID, taskID):
    project_and_task_path = f'project/{projectID}/task/{taskID}'
    with open(f"/webodm/app/media/{project_and_task_path}/assets/images.json", 'r') as file:
        camera_shots = parse_json(file)
        for image_shot in camera_shots:
            ImageUpload.objects.update_or_create(task=Task.objects.get(pk=taskID),
                                                 image=f"{project_and_task_path}/{image_shot['filename']}")
            print(f"Succesfully indexed file {image_shot['filename']}")
### After pasting function - copy and paste input for reading project ID and task - fill the project ID and task id
### like <projectId>|<taskId>. Example 11|8e7fd260-bfd6-4d17-ae42-d00c8ad09370
projectID, taskID = input("Enter the project ID (numeric) and taskID via vertical slash '|' : ").split("|")
### Copy and paste this for starting reindexing images after import
reindex_shots(projectID,taskID)
```

markFieldman, 2021-11-14:
@pierotofy Can we use it as base of the reindexing function inside the WebODM?

jakevis, 2021-12-08:
Sorry for the delay - but this fixed it and works for me. Would be awesome if there was a webui "reindex" button too - but this works! - thanks for all your support.

Zuline, 2024-02-27:
I created a pull request today for the Readme.md showing a simply Bash shell command that adds the images. No manual re-indexing is required.

pierotofy, 2024-05-31:
This is now fixed :clinking_glasses:

closed 2024-05-31 by pierotofy