WebODM/WebODM #872: Monitor active number of workers to notify users of memory issues

pierotofy, 2020-05-30:
If WebODM runs out of memory, the OOM will start killing processes at random. Sometimes if the process killed is the worker process, WebODM will stop functioning properly and no error messages will be displayed to the user. This should be improved.

pierotofy, 2020-06-04:
This seems to be actually more frequently caused when we run out of disk space.

dmbworld, 2020-06-04:
I've noticed this certainly when running low on RAM.

pierotofy, 2025-03-12:
This is a complicated issue. If disk space is exhausted, or low memory causes the OOM to trigger, there's no reliable way to communicate this with the end user. (This is even more complicated if there's a virtualization layer in the middle of it).

closed 2025-03-12 by pierotofy