WebODM/WebODM #873: Point cloud not displaying and no error, but mesh does display

gjsa73, 2020-06-01:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 83 on Windows 10

### What is the problem?

Point clouds are not displaying in WebODM:

- The canvas loads but only shows the selected background and any other features (eg cameras,cube ,etc).
- No error is generated (as has been reported previously.  The last log entries are:

```
90
.
.
.
100
Found point cloud: odm_georeferencing/odm_georeferenced_model.laz
Entwine point cloud is already built.
Postprocessing: done (•̀ᴗ•́)و!
Compressing all.zip
Done!
```

- If I select `Textured Model -> Show Model`, the mesh displays after a reasonable loading time.

### What should be the expected behavior?

Point cloud should display by default

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Do not think this is data specific - have installed WebODM as a Docker instance using the instructions in GitHub and this problem occurs.  If however, I install the native [Windows WebODM](https://webodm.net/static/downloads/WebODM_Setup.exe), and process the same data, the point clouds do display in the canvas.

pierotofy, 2020-06-01:
Hey @gjsa73 :hand: thanks for the report; if you open press CTRL+SHIFT+J (to open the Developer Toolbar), then navigate to the "Console" tab and refresh the page, do you see any error messages?

gjsa73, 2020-06-01:
@pierotofy here is a screenshot of the errors after refreshing the page:
[image]

gjsa73, 2020-06-01:
Found an issue in potree and tried one of the suggestions successfully:

Does it work when you **disable Eye-Dome-Lighting** and set splat-quality to standard?
**Yes**

_Originally posted by @m-schuetz in https://github.com/potree/potree/issues/541#issuecomment-416225053_

pierotofy, 2020-06-02:
That could definitely be done; are you familiar with Javascript? If so would you be interested in contributing a pull request? It can probably be done with some changes here https://github.com/OpenDroneMap/WebODM/blob/master/app/static/app/js/ModelView.jsx#L194

gjsa73, 2020-06-03:
Just updated the title as the problem only affects the latest Chrome and Microsoft Edge.  Firefox v76 displays the point clouds by default with Eye-Dome Lighting enabled.  Will use Firefox

closed 2020-06-03 by gjsa73

trevor-miracad, 2020-06-18:
```
initThree()
let context = canvas.getContext('webgl2', contextAttributes );
```

setting the three.js WebGLRenderer to 'webgl2' causes this issue for me when doubleclicking on the pointcloud.  Very similar error:  

```
viewer.js:879 Uncaught could not compile shader pointcloud: WARNING: 0:7: 'GL_EXT_frag_depth' : extension is not supported
ERROR: 0:74: 'gl_FragDepthEXT' : undeclared identifier
ERROR: 0:74: 'assign' : l-value required (can't modify a const)
```

To fix it, ensure  the context is set to "webgl"
Thought I would share my findings - if set, I have a feeling it would fix this error also

commit 06c4dbd7 (2020-06-04): Fix: #873
commit 97a7a8af (2020-06-04): Merge pull request #874 from pierotofy/camerajson

Fix: #873