WebODM/WebODM #877: Contours not working on fresh install

ivanmalo, 2020-06-06:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 83 on Windows 10

### What is the problem?

I created a task with 545 images (although any task has the same issue). The process went fine and DSM is enabled. 

When I try to preview Contours I get: "Invalid GeoJSON object.". 
When I try to export I get: 

HTTP 200 OK
Allow: GET, HEAD, OPTIONS
Content-Type: application/json
Vary: Accept

{
    "error": "Invalid task output (cannot find valid key)"
}

### What should be the expected behavior?

Countours previewed or GeoJSON exported.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Fresh native install with bought install script for Ubuntu 16.04. Create task with DJI images and try to preview Contours. We do get Plant Health, Surface Model and 3D model working fina.

ivanmalo, 2020-06-18:
Is there any update? Can I do something to investigate further? I really have tried everything and can't find a clue...

pierotofy, 2020-06-18:
Check the logs from the `worker` container with `docker logs webodm_worker`. I suspect there's something wrong with the installation of GRASS since this is a native installation?

Does it work if you use the docker version? (Install procedure: https://docs.opendronemap.org/installation.html#linux)

ivanmalo, 2020-07-08:
I get no information from that command. Although I saw this when I started a Celery Worker:

ERROR Could not execute GRASS script /webodm/plugins/contours/calc_contours.py from /webodm/app/media/tmp/tmpejnopl4q_grass_engine: Starting GRASS GIS...
Creating new GRASS GIS location <location>...
Executing <python /webodm/plugins/contours/calc_contours.py dem_file=/webodm/app/media/project/11/task/7243fb22-2a67-47e3-96d6-4cb607f7fdd0/assets/odm_dem/dsm.tif epsg=43
26 simplify=0.2 format=GPKG interval=1.0> ...
Traceback (most recent call last):
  File "/webodm/plugins/contours/calc_contours.py", line 92, in <module>
    sys.exit(main())
  File "/webodm/plugins/contours/calc_contours.py", line 52, in main
    if opts['format'] == "GeoJSON":
KeyError: 'format'
Execution of <python /webodm/plugins/contours/calc_contours.py dem_file=/webodm/app/media/project/11/task/7243fb22-2a67-47e3-96d6-4cb607f7fdd0/assets/odm_dem/dsm.tif epsg
=4326 simplify=0.2 format=GPKG interval=1.0> finished.
Cleaning up temporary files...
[2020-07-08 20:40:39,528: ERROR/ForkPoolWorker-3] worker.tasks.execute_grass_script[3592a118-202f-4e9f-a23d-18edf174a4b9]: Could not execute GRASS script /webodm/plugins/
contours/calc_contours.py from /webodm/app/media/tmp/tmpejnopl4q_grass_engine: Starting GRASS GIS...
Creating new GRASS GIS location <location>...
Executing <python /webodm/plugins/contours/calc_contours.py dem_file=/webodm/app/media/project/11/task/7243fb22-2a67-47e3-96d6-4cb607f7fdd0/assets/odm_dem/dsm.tif epsg=43
26 simplify=0.2 format=GPKG interval=1.0> ...
Traceback (most recent call last):
  File "/webodm/plugins/contours/calc_contours.py", line 92, in <module>
    sys.exit(main())
  File "/webodm/plugins/contours/calc_contours.py", line 52, in main
    if opts['format'] == "GeoJSON":
KeyError: 'format'
Execution of <python /webodm/plugins/contours/calc_contours.py dem_file=/webodm/app/media/project/11/task/7243fb22-2a67-47e3-96d6-4cb607f7fdd0/assets/odm_dem/dsm.tif epsg
=4326 simplify=0.2 format=GPKG interval=1.0> finished.
Cleaning up temporary files...

pierotofy, 2021-02-05:
Hey @ivanmalo :hand: this should have been fixed in the newest version of WebODM. Please test and if the error is still there re-open this issue? Thanks!

closed 2021-02-05 by pierotofy