WebODM/WebODM #878: Uncaught ReferenceError: THREE is not defined

csotomon, 2020-06-07:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 83 on macOS (Catalina)

### What is the problem?

I am using WebODM with dockers, and using the 2D and 3D views iframes from an ReactJS page.

I activated SSL in WebODM, and when i open the ReactJS view, the 2D map view works ok, but the 3D view have rendering problems and in the browser console i can see this error message.

`  
ModelView-a1845963214113052217.js:14 Uncaught ReferenceError: THREE is not defined
    at Object.9 (ModelView-a1845963214113052217.js:14)
    at s (ModelView-a1845963214113052217.js:1)
    at Module.156 (ModelView-a1845963214113052217.js:6)
    at s (ModelView-a1845963214113052217.js:1)
    at Object.155 (ModelView-a1845963214113052217.js:6)
    at s (ModelView-a1845963214113052217.js:1)
    at ModelView-a1845963214113052217.js:1
    at ModelView-a1845963214113052217.js:1
`
Everything works fine with out SLL but with SLL the 3D rendering crashes

### What should be the expected behavior?

the 3D map rendering into an iframe in a ReactJS page using SSL

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Create a ReactJS APP with a 3D WebODM iframe, and both (ReactJS and WebODM docker version) working with SSL.

You can see that de 3D iframe crash but the 2D iframe works fine.

pierotofy, 2020-06-07:
Hey @csotomon :hand: are you using the latest version (with this fix included? https://github.com/OpenDroneMap/WebODM/pull/875

csotomon, 2020-06-07:
Hi @pierotofy . Yes i am.

pierotofy, 2020-06-07:
Do you have a public link we could check to reproduce this?

csotomon, 2020-06-07:
Sure, i created a reactjs web page for testing, with the 2D and 3D view iframes in the same page.

Here you can see that the 2d view works, but the 3d view doesn't render.

[https://webodmtest.azurewebsites.net/](https://webodmtest.azurewebsites.net/)

pierotofy, 2020-06-08:
Thanks for the link!

This was indeed a bug. Some tags were not being properly included due to recent changes we made to the 3D model display.

Please update and verify if it works; if not, please re-open this. :+1:

closed 2020-06-08 by pierotofy

csotomon, 2020-06-08:
@pierotofy , thank you for your help. I can't test it because i'm using the docker version, and the image is not updated in dockerhub, can U upload the new image?

pierotofy, 2020-06-08:
It should be updated by now (builds are automatic, but take a few hours to show up).

csotomon, 2020-06-08:
It's woking, thank you

PR #879 (2020-06-08): Fix 3D iframe display
Closes #878