WebODM/WebODM #880: Problem in downloading WebODM

alimadadmsbk, 2020-06-09:
****************************************

Greetings and thank you for the implementation of this precious product.
I work in a photogrammetry office at the National Cartography Center of Iran and I always try to orient the organization towards open source technologies. I hope it is possible to popularize this system in the organization too.
I am starting to install WebODM in the Windows 7 environment. The Git, Python, and Docker are installed, and the values of memory and processor are defined too. Now I am downloading WebODM. 
When I launch Git Gui and set below address as the source: "https://github.com/Open-DroneMap/WebODM", the following message is received:
fatal: unable to access "https://github.com/Open-DroneMap/WebODM/": connection to port 443 of github.com failed: timeout exceeded.
Adding that when I go directly to this address via Chrome, the 404 error is reported. The same message is delivered without a proxy or with a VPN.
I will be very happy to receive advice on how to resolve this issue and proceed with the installation process.
Thank you very much for your time.
Sincerely, Ali Madad

pierotofy, 2020-06-09:
Hi @alimadadmsbk,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-06-12 00:20) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

alimadadmsbk, 2020-06-09:
Thanks a lot
Excuse me. I made a mistake and will move my question to the Community Forum.
Best Regards
Ali Madad

closed 2020-06-09 by pierotofy