WebODM/WebODM #883: Permanent Settings for Custom Basemaps

bearyjd, 2020-06-18:
I have a feature request for folks that are working in an offline environment. Would it be possible in the 2D map rendering engine to allow preference to be saved with respect to the custom basemap?

When you enter a custom basemap, it correctly displays, but does not survive a refresh and it defaults back to google.

I humbly suggest that you allow this entire list to be customized somehow – either in settings (site wide) or even user by user. This would preferably allow multiple custom basemaps (satellite, openstreetview tile server, etc). When working in low bandwidth environments, I have found that bringing a few TBs of map data is not too terrible and i just run my own tiling servers.

Thanks again for an amazing product. the community behind webodm is amazing.

pierotofy, 2020-06-18:
Hi @bearyjd,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-06-21 06:35) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2022-10-19:
This should be improved with https://github.com/OpenDroneMap/WebODM/commit/446fdb798742bc2693f28d2505ee6762bf26166f as the program will now remember the last basemap used.

We'd welcome a PR for the ability to manage more basemaps.

closed 2022-10-19 by pierotofy