WebODM/WebODM #884: Scaling issue

FazioliPhotography, 2020-06-23:
Hi! I processed a building facade 3D model with 93 UAV images, half taken at 90 degrees and then repeated at 75. Overlap 78%.

The point cloud looks amazing. Absolutely gorgeous. Problem is, it measures about 92% of the actual size of the building. What happened, and how do I fix it?

The images were generated by a third party, because we do remote modeling.

pierotofy, 2020-06-23:
Hi @FazioliPhotography,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-06-26 09:20) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

closed 2020-06-24 by pierotofy

smathermather, 2020-06-25:
This would be a great question on Community.opendronemap.org, if you haven't already posted there.