WebODM/WebODM #885: How to migrate new model webodm

boopathit98, 2020-06-27:
Hi there ,
Please help me out. For create a new model and migrate not working i will try lot not working . i will create a new table as request_image for separate . how to solve please help me.

python manage.py migrate
 not working

pierotofy, 2020-06-27:
Hi @boopathit98,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-06-30 02:45) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

boopathit98, 2020-06-27:
[image]
sudo docker-compose run migration
WARNING: Found orphan containers (webodm_node-odm_1) for this project. If you removed or renamed this service in your compose file, you can run this command with the --remove-orphans flag to clean it up.
Starting db ... done
python: can't open file 'manage.py': [Errno 2] No such file or directory

boopathit98, 2020-06-27:
[image]
services:
  db:
    image: opendronemap/webodm_db
    container_name: db
    ports:
      - "5432"
    volumes:
      - dbdata:/var/lib/postgresql/data:Z
    restart: unless-stopped
    oom_score_adj: -100
  migration:
    image: opendronemap/webodm_db
    command: python manage.py migrate --database=users
    volumes:
      - dbdata:/var/lib/postgresql/data:Z
    depends_on:
      - db

boopathit98, 2020-06-27:
migrations.CreateModel(
            name='RequestImage',
            fields=[
                ('id', models.AutoField(auto_created=True, primary_key=True, serialize=False, verbose_name='ID')),
		('project', models.ForeignKey(to='app.Project', on_delete=django.db.models.deletion.CASCADE, help_text="Project this image belongs to")),
                ('image', models.ImageField(help_text='File uploaded by a user', upload_to=app.models.image_directory_path)),
		('status', models.IntegerField(blank=True,  db_index=True, help_text='Current status of the task', null=True)),
            ],
        ),

closed 2020-06-27 by pierotofy