WebODM/WebODM #887: Missing documentation for [POST]/tasks/import endpoint

RGrze, 2020-07-07:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 78 on Ubuntu Linux

### What should be the expected behavior?

There is no documentation for `[POST] /api/projects/{project id}/tasks/import` endpoint at [WebODM Documentation webpage](https://docs.webodm.org/#task)

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

..

pierotofy, 2020-07-07:
Hey @RGrze :hand: see https://github.com/OpenDroneMap/WebODM/blob/5ad0ccc764016a6f2ba1d29c74c663e9d3e0d882/app/api/tasks.py#L360

Would you be interested in opening a PR for this?

RGrze, 2020-07-07:
@pierotofy hey, sure, I'm on it :)

closed 2020-07-07 by pierotofy