WebODM/WebODM #896: DSM-DTM Elevation calculations inconsistent and incorrect

dmbworld, 2020-08-03:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

whatismybrowser.com/w/SRDHD5P

### What is the problem?

Recalculation of an image set using new versions of WebODM produce incorrect elevation measurements.

### What should be the expected behavior?

Elevation calculations should be the same if not very similar.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Tough one. Try processing older image sets in newer version of WebODM and comparing?

I have attached a PDF file with the screen shots and short explanations
[DSM-DTM-Elevation-Incorrect.pdf](https://github.com/OpenDroneMap/WebODM/files/5017583/DSM-DTM-Elevation-Incorrect.pdf)

pierotofy, 2020-08-03:
Could you share the `dsm.tif` file from both datasets?

Seems like a histogram calculation issue.

dmbworld, 2020-08-03:
I've emailed a OneDrive link to your "masseranolabs.com" email, let me know if it warrives.

pierotofy, 2020-08-03:
Mm, the only change that could have affected this is the new parallel filterpoints. https://github.com/OpenDroneMap/ODM/commit/7947df0787f9dc8ce4bd28973e90d8509ec8d3c5

Could you try to reprocess this, but setting `max-concurrency` to `1` (which disables the parallel filterpoints implementation).

dmbworld, 2020-08-03:
Just changed the setting and restarted the task, will take around 3hr:44min. I'll let you know the result as soon as I wake. This may explain another random error where 3D models are randomly flipped differently for each run of the same image data.

dmbworld, 2020-08-04:
Here is the result after setting max-concurrency to 1:
[image]

I have upload the second version of the dsm to the same link as before and it is called old-dsm-2.tif
As you will see the change is now in the negative elevation direction completely so clearly something else is affecting this.

pierotofy, 2020-08-04:
Mm, indeed. I'm not really sure then. Note that the majority of the model itself looks OK, it's that there seem to be a few outliers in the model that push the range of the min/max values in the histogram.

dmbworld, 2020-08-04:
Setting max-concurrency to 1 did make a difference and the range changed from -2578.148 / 182.301 and instead gave -9735.19/62666.797.
There is certainly some processing differences for the numbers to change and back in december 2019 the results were feasible and all these tests are using the same dataset. I'm not familiar enough with the code that is used to suggest any possible calculation differences. The idea of the parallel processing having some effect made sense until we set the value to 1. I've tested many of the old datasets now and they all give really unusual results with this version. I suspect some randomness so I'll test the same dataset with these settings again.

dmbworld, 2020-08-05:
When checking the contours they are identical and good showing the correct heights in all cases of the same dataset rerun at different times so using different versions of the ODM modules. The problem appears to be with the DSM and DTM ranges. There is some calculation problem there which have been introduced over the past 7 months or so.
@pierotofy would it help you if I PM a shared link to the server where the differing models using the same dataset are and this way you can probably spot what is happening?

pierotofy, 2020-08-06:
A link to the point cloud file would help!

dmbworld, 2020-08-06:
> 
> 
> A link to the point cloud file(s) would help!

@pierotofy, Email sent with the links.

pierotofy, 2020-08-06:
Verdict:
 - Nothing is wrong with the files; the results are correct (both point cloud and DSM)
 - This looks like a bug in `rio_tiler`'s `_stats` function, triggered by some odd behavior by `WarpedVRT` (replacing `WarpedVRT` with a plain rasterio dataset yields correct min/max results for the histogram).

A straightforward fix is to reverse rasterio to 1.1.0 and GDAL to 2.1, expect that would nullify the work we've done to upgrade GDAL and take advantage of faster/stable COG generation. Since there's nothing wrong with that, this feels like a step backward.

This is going to be a tricky one to fix and requires way more time than I have available currently. Hopefully somebody can provide funding and/or a workaround.

pierotofy, 2020-08-06:
I should add, this might  be caused by GDAL creating overviews with erroneous values (the histogram is computed on one of the overviews).

pierotofy, 2020-08-06:
Possibly related: https://github.com/OSGeo/gdal/issues/2671

pierotofy, 2020-08-06:
We might be lucky and the fix is to make sure we sample overviews using a nearest filter (matches previous `cog_translate` behavior).

closed 2020-08-06 by pierotofy

dmbworld, 2020-08-08:
@pierotofy an update: I have now re-run 6 control tests with the older datasets and they compare correctly with original results so this appears to be the fix. Thanks for spotting the source of the problem and applying the fix.

pierotofy, 2020-08-08:
Glad it's working! Thanks for testing and reporting back.

PR #899 (2020-08-06): Resample overviews using nearest filter
This should fix #896