WebODM/WebODM #900: Contours works on server but nowhere else CSFR verification failed. Please help

adamglee, 2020-08-19:
What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[https://www.whatismybrowser.com/w/WX698BF?utm_source=whatismybrowsercom&utm_medium=internal&utm_campaign=homepage&utm_content=unique-url-readout]  Chrome 64 bit ( hope i've done this correctly

### What is the problem?

Contours works properly on my server with webodm installed but as soon as I try on any other computer in or outside of the network I get this error when clicking on the preview button to bring up the contours for any and all maps. 

( {"readyState":4,"responseText":"\n<!DOCTYPE html>\n<html lang=\"en\">\n<head>\n <meta http-equiv=\"content-type\" content=\"text/html; charset=utf-8\">\n <meta name=\"robots\" content=\"NONE,NOARCHIVE\">\n <title>403 Forbidden</title>\n <style type=\"text/css\">\n html * { padding:0; margin:0; }\n body * { padding:10px 20px; }\n body * * { padding:0; }\n body { font:small sans-serif; background:#eee; color:#000; }\n body>div { border-bottom:1px solid #ddd; }\n h1 { font-weight:normal; margin-bottom:.4em; }\n h1 span { font-size:60%; color:#666; font-weight:normal; }\n #info { background:#f6f6f6; }\n #info ul { margin: 0.5em 4em; }\n #info p, #summary p { padding-top:10px; }\n #summary { background: #ffc; }\n #explanation { background:#eee; border-bottom: 0px none; }\n </style>\n</head>\n<body>\n<div id=\"summary\">\n <h1>Forbidden <span>(403)</span></h1>\n <p>CSRF verification failed. Request aborted.</p>\n\n\n <p>You are seeing this message because this site requires a CSRF cookie when submitting forms. This cookie is required for security reasons, to ensure that your browser is not being hijacked by third parties.</p>\n <p>If you have configured your browser to disable cookies, please re-enable them, at least for this site, or for &#39;same-origin&#39; requests.</p>\n\n</div>\n\n<div id=\"explanation\">\n <p><small>More information is available with DEBUG=True.</small></p>\n</div>\n\n</body>\n</html>\n","status":403,"statusText":"Forbidden"}
Interval:
Layer:
Simplify:
Projection:
+
−
500 m
1000 ft
-17.52587 : 145.91384 )

### What should be the expected behavior?

With this tab(contours )on work server it works gI

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Contours Tab with above error (cut & paste)

this is the address to the server anyone is more than welcome to try it  http://101.167.37.202:8000/public/task/d5a89e62-3a69-4323-8f65-bd6e2074fbfe/map/

Help is appreciated and am very thankful to anyone who can help as this is the one setting I need for my client .  I have installed and re-installed WebODM and also Docker. I have opened 8000 port on my router and also tried and installed WebODM on my laptop ( 10 Images as only 8gb ram) and also if open it on laptop to look at contours through edge i can replicate the error ( chrome on laptop works )

pierotofy, 2020-08-19:
Can confirm, this is a bug.

pierotofy, 2020-08-19:
This should be fixed. Please wait a few hours until the docker images get rebuilt, update and test to see if it's solved? Thanks!

closed 2020-08-19 by pierotofy

adamglee, 2020-08-19:
Thank you for all your help 

Can you guide me through how to update images as last time I tried to update docker I lost all my jobs. Do u simply push the update button in docker or is it in the webodm interface 

Cheers mate 👍

Adam

pierotofy, 2020-08-19:
I guess it depends on how you installed WebODM.

If you bought the installer, just press "Update" from WebODM Manager.

If you installed via command line, run `./webodm.sh update`

adamglee, 2020-08-19:
I've just gotten this error in Webodm console trying to update.  Help is appreciated

 docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.nodemicmac.yml down --remove-orphans
Removing network webodm_default
Network webodm_default not found.
Updating WebODM...
git pull origin master
From https://github.com/OpenDroneMap/WebODM
 * branch            master     -> FETCH_HEAD
Already up-to-date.
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml pull
Pulling db       ... 
Pulling broker   ... 
Pulling worker   ... 
Pulling node-odm ... 
Pulling webapp   ... 
[5A[2K
Pulling db       ... [31merror[0m
[5B[4A[2K
Pulling broker   ... [31merror[0m
[4B[1A[2K
Pulling webapp   ... [31merror[0m
[1B[2A[2K
Pulling node-odm ... [31merror[0m
[2B[3A[2K
Pulling worker   ... [31merror[0m
[3B
ERROR: for db  Get https://registry-1.docker.io/v2/: Bad Request
ERROR: for broker  Get https://registry-1.docker.io/v2/: Bad Request
ERROR: for webapp  Get https://registry-1.docker.io/v2/: Bad Request
ERROR: for node-odm  Get https://registry-1.docker.io/v2/: Bad Request
ERROR: for worker  Get https://registry-1.docker.io/v2/: Bad Request
Get https://registry-1.docker.io/v2/: Bad Request
Get https://registry-1.docker.io/v2/: Bad Request
Get https://registry-1.docker.io/v2/: Bad Request
Get https://registry-1.docker.io/v2/: Bad Request
Get https://registry-1.docker.io/v2/: Bad Request
Error: Update exited with code: 1

adamglee, 2020-08-19:
Hi Piero 

I have followed required steps and get this error 

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.nodemicmac.yml down --remove-orphans
Removing network webodm_default
Network webodm_default not found.
Updating WebODM...
git pull origin master
From https://github.com/OpenDroneMap/WebODM
 * branch            master     -> FETCH_HEAD
Already up-to-date.
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml pull
Pulling db       ... 
Pulling broker   ... 
Pulling worker   ... 
Pulling node-odm ... 
Pulling webapp   ... 
[5A [2K
Pulling db       ... [31merror [0m
[5B [4A [2K
Pulling broker   ... [31merror [0m
[4B [1A [2K
Pulling webapp   ... [31merror [0m
[1B [2A [2K
Pulling node-odm ... [31merror [0m
[2B [3A [2K
Pulling worker   ... [31merror [0m
[3B
ERROR: for db  Get https://registry-1.docker.io/v2/: Bad Request
ERROR: for broker  Get https://registry-1.docker.io/v2/: Bad Request
ERROR: for webapp  Get https://registry-1.docker.io/v2/: Bad Request
ERROR: for node-odm  Get https://registry-1.docker.io/v2/: Bad Request
ERROR: for worker  Get https://registry-1.docker.io/v2/: Bad Request
Get https://registry-1.docker.io/v2/: Bad Request
Get https://registry-1.docker.io/v2/: Bad Request
Get https://registry-1.docker.io/v2/: Bad Request
Get https://registry-1.docker.io/v2/: Bad Request
Get https://registry-1.docker.io/v2/: Bad Request
Error: Update exited with code: 1 Thank you again your a legend 👍

Adam

adamglee, 2020-08-20:
Resolved , had my proxy setup, removed and updated will let you know if fixes my issue ..... Once again thank you mate

Adam

adamglee, 2020-08-20:
Thank you 🙏 

All working ... contours now working correctly though work server and also on clients system. 

Cheers 

Adam

PR #901 (2020-08-19): Ensure CSRF cookie on public views
Fixes #900