WebODM/WebODM #902: Duplicate layer list in map viewer for selected project

gjsa73, 2020-08-20:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 84 on Windows 10
Edge 84 on Windows 10

### What is the problem?

When viewing a project that contains multiple tasks, the individual tasks are loading in the task list with many duplicates.  See below:

[image]

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Only one unique task per output should be listed

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Unsure

gjsa73, 2020-08-21:
Update: onyl the labels are duplicates, not the actual orthophotos.  In other words - the correct number of tasks are listed for the project - and the orthophotos displayed are unique - but the task names are incorrect/duplicated.

gjsa73, 2020-08-25:
@pierotofy if you can direct me to where the layers-control-panel -> layers-control-layer -> layer-label class is populated and I will troubleshoot why the labels do not match each layer.

pierotofy, 2020-12-16:
Thanks for the report! :pray: 

This should be fixed with https://github.com/OpenDroneMap/WebODM/pull/943/commits/f99ee751710a27df7689670d7e36ad59546a7573 which will be included in the next release.

pierotofy, 2020-12-23:
Please update, see if the problem is solved and reopen if the issue persists? :pray:  Thanks!

closed 2020-12-23 by pierotofy

gjsa73, 2021-01-04:
Can confirm the issue has been resolved, thanks! We need a scroll bar on
the layer list panel but I'll open another issue for that.