WebODM/WebODM #903: Elevation not working on windows 10, Docker 2.0.0.3 (31259)

adamglee, 2020-08-21:
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

https://www.whatismybrowser.com/w/JTK5RTF?utm_source=whatismybrowsercom&utm_medium=internal&utm_campaign=homepage&utm_content=unique-url-readout

My server is i5 , 32gb ram , windows 10. running up to date WebODM and docker is 2.0.0.3 (31259)

### What is the problem?

Elevation not working I have looked around and cant find a answer to this, apologizes if so.  all other plugins work great and contours works well now thanks to Piero Toffanin for the help. this is the error(cut and paste)

Could not execute GRASS script /webodm/plugins/elevationmap/elevationmap.py from /webodm/app/media/tmp/tmpa4q37o76_grass_engine: Starting GRASS GIS... Creating new GRASS GIS location <location>... Cleaning up temporary files... Executing <python /webodm/plugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/4/task/d5a89e62-3a69-4323-8f65-bd6e2074fbfe/assets/odm_dem/dsm.tif interval=5 format=GeoJSON noise_filter_size=0.0 epsg=4326> ... Traceback (most recent call last): File "/webodm/plugins/elevationmap/elevationmap.py", line 48, in <module> import cv2, math, argparse ModuleNotFoundError: No module named 'cv2' Execution of <python /webodm/plugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/4/task/d5a89e62-3a69-4323-8f65-bd6e2074fbfe/assets/odm_dem/dsm.tif interval=5 format=GeoJSON noise_filter_size=0.0 epsg=4326> finished. Cleaning up temporary files...

### What should be the expected behavior?

?Would like to use the elevation plugin as would be of great value to myself.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? Please include multiple screenshots of the problem! If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I simply click the plugin button on the right and this error comes about. I have tried to change some settings as seem in another post ( do not filter noise 0m ) in the option for noise filter ( didn't fix it for me)

thank you for reading and hope someone is able to help.  great team and software thank you for all the hard work

closed 2020-10-21 by pierotofy

pierotofy, 2020-10-21:
Should be fixed with #927. Please update and re-open if the issue persists? pray

adamglee, 2020-10-27:
Thank you for the hard work mate 👍

Will update webodm and cross my fingers and give it a go. 

Thanks again 

Adam

PR #927 (2020-10-21): Use Ubuntu 18.04 as base image, fix plugins
Solves some build issues reported in #926

Should fix #903 and and #917