WebODM/WebODM #910: Error on rendering 3D Maps

rawi68, 2020-09-15:
I have installed webodm on an ubuntu 20.04  Docker

The opencv python version is 4.2.0 which causes the error below

INFO Booting WebODM 1.4.3

INFO Registered [plugins.cesium-ion.plugin]

INFO Registered [plugins.diagnostic.plugin]

INFO Registered [plugins.contours.plugin]

****INFO Installing requirements.txt for [plugins.changedetection.plugin]

Collecting geojson==2.4.1

  Using cached geojson-2.4.1-py2.py3-none-any.whl (15 kB)

ROR: Could not find a version that satisfies the requirement opencv-python==4.1.0.25 (from -r requirements.txt (line 2)) (from versions: 3.4.8.29, 3.4.9.31, 3.4.9.33, 3.4.10.35, 3.4.10.37, 3.4.11.39, 3.4.11.41, 4.1.2.30, 4.2.0.32, 4.2.0.34, 4.3.0.36, 4.3.0.38, 4.4.0.40, 4.4.0.42)

ERROR: No matching distribution found for opencv-python==4.1.0.25 (from -r requirements.txt (line 2))

WARNING Failed to install requirements.txt for [plugins.changedetection.plugin]****

Any solutions to solve this problem?
Thank you.

pierotofy, 2020-09-15:
Hi @rawi68,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2020-09-18 07:40) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2020-09-19:
Since apparently some of the required information is still missing, I'm closing this now, sorry. Feel free to reopen this or create a new issue once you can provide **all** required information.

closed 2020-09-19 by pierotofy