WebODM/WebODM #911: WebOdm Manager on Mac

farcasclaudiu, 2020-09-18:
****************************************

### What's your browser and operating system?

Chrome 85 on macOS (Catalina)

### What is the problem?

Issue described here
[https://community.opendronemap.org/t/webodm-manager-wont-talk-to-docker/5398/4](https://community.opendronemap.org/t/webodm-manager-wont-talk-to-docker/5398/4)

On Mac OS, after upgrading Docker to version 2.3.0.5, WebODM Manager throws a Fatal error “Docker is already running”. The WebODM Manager says “Docker is installed but not communicating…” and “Launching Docker…” but after a while if it trying to load, throws another error saying “We had some issues launching Docker. Find the Docker application and start it manually, then restart WebODM Manager.”
The issue is that I cannot update webodm images, otherwise I can run old containers manually from docker dashboard.
On Windows even after I upgraded Docker to version 2.3.0.5, WebOdm Manager works just fine and does not complain.

### What should be the expected behavior?

No errors.

### How can we reproduce this?

On Mac, just start WebOdm Manager.

pierotofy, 2020-09-18:
Related: https://community.opendronemap.org/t/webodm-manager-wont-talk-to-docker/5398/5

pierotofy, 2020-09-18:
This is a bug in the latest version of docker.

You can downgrade the docker version or turn off the “Cloud experience” option from the Docker Preferences. https://github.com/docker/for-mac/issues/4919

[image]

farcasclaudiu, 2020-09-21:
The workaround works indeed.
Thank you for pointing to the solution.
Best regards!

closed 2020-09-21 by farcasclaudiu